SpacehuhnTech/esp8266_deauther #1: can't get it working on esp 01

luisrx7, 2017-02-03:
Hi
I ulpoaded the code to my esp 01 via a usb to serial converter and the esp does not create the AP and do not send "Starting" to the serial port. what am I doing wrong?

spacehuhn, 2017-02-04:
That's hard to say. Do you get an error message when uploading? My ESP-01 works fine with the code.

luisrx7, 2017-02-05:
how do you supply power to the esp?

spacehuhn, 2017-02-05:
Please stay at the issue..
I used a USB to serial adapter for programming and as 3.3V powersupply.

luisrx7, 2017-02-05:
that is exacly what I'm using too. could you send me the bin file so I can flash it to my esp?

spacehuhn, 2017-02-05:
I don't have a bin file. You can compile and upload via Arduino.

luisrx7, 2017-02-05:
I know. I'm asking you to compile your code with your settings and send me the bin file.

mrx23dot, 2017-03-12:
v2.0.0 sdk needs min 8Mbit ROM, ESP-01 only has 4Mbit max
use NodeMCU 32Mbit

mrx23dot, 2017-03-12:
Would it be possible to add support of SDK 1.4.0 which is compatible with ESP-01 4Mbit cheapest module, (for eg. mass deployment)?

spacehuhn, 2017-03-12:
### Uploading the bin files  

**Note:** the 512kb version won't have the full MAC vendor list.  

**0** Download the current release from [here](https://github.com/spacehuhn/esp8266_deauther/releases)  

**1** Upload using the ESP8266 flash tool of your choice. I recommend using the [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher). If this doesn't work you can also use the official [esptool](https://github.com/espressif/esptool) from espressif.

**That's all! :)**

Make sure you select the right com-port, the right upload size of your ESP8266 and the right bin file.

mrx23dot, 2017-03-12:
(sorry size limit was only for AT firmware)

But still it doesnt work on ESP-01:
python C:\Python34\Scripts\esptool.py -p COM9 -b 115200 write_flash 0x00000 esp8266_deauther_1.1_512kb.bin

esptool.py v1.3
Connecting....
Auto-detected Flash size: 4m
Running Cesanta flasher stub...
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.5 kbit/s)...
Leaving...

remove GPIO0 pull down, Reset
Only PWR LED is lit, no UART communication out, no new AP name.
Is my 0x00 flash address okay?

Anikdesign, 2017-03-19:
8MB Board  Here is the Errors I get . Any Ideas ??

ESP8266WiFi_1.cpp:564:8: error: prototype for 'int8_t ESP8266WiFiClass::scanNetworks(bool)' does not match any in class 'ESP8266WiFiClass'
 int8_t ESP8266WiFiClass::scanNetworks(bool async)

SP8266WiFi.h:243:12: error: candidate is: int8_t ESP8266WiFiClass::scanNetworks(bool, bool)

     int8_t scanNetworks(bool async = false, bool show_hidden = false);

            ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

spacehuhn, 2017-03-19:
readme...

12 Go to the SDK_fix folder of this project

13 Copy ESP8266WiFi.cpp and ESP8266WiFi.h

14 Past these files here packages > esp8266 > hardware > esp8266 > 2.0.0 > libraries > ESP8266WiFi > src

mrx23dot, 2017-03-19:
Hi spacehuhn, have you tested the 512.bin on ESP-01 4Mbit? Did it work for you?

If so could you send me how did you flash it? Maybe I used bad address:
python esptool.py -p COM9 -b 115200 write_flash 0x00000 esp8266_deauther_1.1_512kb.bin

spacehuhn, 2017-03-19:
The only ESP-01 that I have has 1mb and I have no problem flashing it. Your settings seem to be fine. I can't say why it's not working.
That's why I recommend using a USB dev board, way less stress with these.

sirjeannot, 2017-03-19:
Maybe not worth a new topic, but it crashes on an esp8266 esp-201. I've tried all combinations, I get stack dumps and reboots.

mrx23dot, 2017-03-22:
Got it to work on ESP-01 4Mbit have to compile with: 512KB (64K SPIFFS)

[esp8266_deauther_esp01.zip](https://github.com/spacehuhn/esp8266_deauther/files/862639/esp8266_deauther_esp01.zip)

Vaibhavland, 2017-03-28:
Plz Give me. The Bin file for Esp8266 normal edition 4mb one plzz give me i cannot compile this code its giving me alot of errors and even my board is also not showing any wifi signals after this i m helpless plzz help me plz give me the bin file! Plzzz give me

mrx23dot, 2017-03-28:
attached it zipped

Vaibhavland, 2017-03-28:
Is the esp01 the normal esp 8266 we get? The one is having golden lining at the top?

spacehuhn, 2017-03-31:
Can you test if the new realease works for you?
https://github.com/spacehuhn/esp8266_deauther/releases/tag/v1.2

mrx23dot, 2017-03-31:
esp8266_deauther_1.2_512kb.bin works great on ESP01 4mbit!
(had to try flashing it multiple times, bootloader sucks)
Issue can be closed, thank you!

closed 2017-03-31 by spacehuhn