SpacehuhnTech/esp8266_deauther #4: Deauth not working on WeMos D1Mini

panagiks, 2017-02-04:
Hi,

I've installed the the tool on a WeMos D1Mini (Esp-8266 based) and everything works fine except deauthentication. It manages to correctly scan for surrounding APs and finds all the connected clients to a given AP. After that Beacon Spam and Random Beacon Spam works perfectly but when trying to use either of the two deauthentication methods (targeted or to all) under "Status" it displays 0pkts/s and there is no visible effect on any client.

spacehuhn, 2017-02-04:
This sounds like an issue with the SDK not the chip itself. 
I have tested the code on several wemos boards and never had a problem. 

Have you followed all installation steps correctly? 
Because if you don't use the right SDK version the deauthentication attack will be blocked by the chip itself. That's why you need to use version 2.0.0 and edit the user_interface.h file. Only then it's possible to send these packets.

panagiks, 2017-02-04:
Ah fail .... my bad. Although I did follow the installation instructions and installed the 2.0.0 SDK I also had manually installed another version of the SDK for WeMos and was choosing that from the boards drop list. Changed it and deauth worked like a charm. 
Sorry for the false flag and cheers for the great tool!

closed 2017-02-04 by panagiks