SpacehuhnTech/esp8266_deauther #5: Scan wont pick up APs

nutterdk, 2017-02-06:
Im using this board:
http://www.ebay.co.uk/itm/NEW-NodeMcu-DevKit-ESP8266-12E-4MB-WIFI-CP2102-USB-Internet-of-Things-IOT-/231983263238?hash=item360347ae06:g:CVkAAOSwnipWU62m&clk_rvr_id=1162855750654&rmvSB=true

It uploaded and works fine. except when scanning for networks it just keeps scanning and doesnt find anything. Even when its right next to the router

spacehuhn, 2017-02-06:
Have you tried to upload one of the examples and are they working? There is one example (for the ESP8266) for scanning APs which prints them out in the serial monitor.
Please try that out and tell me if this works.

nutterdk, 2017-02-06:
Could you link me to the example?

nutterdk, 2017-02-06:
And by the way its the same card that Seytonic used in his video

spacehuhn, 2017-02-06:
If you followed the steps correctly you should have the SDK installed in Arduino and with that the examples.
So under tools select an ESP8266 and then go to `File `-> `Examples `-> `ESP8266WiFi `-> `WiFiScan`.

nutterdk, 2017-02-06:
Sorry I'm new to arduino. I've uploaded the wifiscan to the nodemcu, now what? How do i test this example

nutterdk, 2017-02-06:
I managed to work that out, so after uploading the wifiscanner example it just spits out all this random text "¢ÿÀ)óÿVÿå1üþDà2ÿÿô��ùú¢ÈBÿÿøå%þü*ðÿÿù"

spacehuhn, 2017-02-06:
Go to `Tool `and open `Serial Monitor` make sure the baud rate (bottom right in the corner) is set to 115200 and then when running the example you should see the scan results  in the monitor.

nutterdk, 2017-02-06:
Ah yeah, it works! It gives a list of the access points near me.

spacehuhn, 2017-02-06:
learned something :D

Now when you upload the deauther code does it work? You can let the serial monitor open and see what it does, maybe you find a clue why it's not working.

nutterdk, 2017-02-06:
Should i reset my board first before uploading the deauther?

spacehuhn, 2017-02-06:
No just upload.

nutterdk, 2017-02-06:
It won't let me.
The Error message and also the port in tools is now greyed out.
Arduino: 1.8.1 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

java.io.IOException: jssc.SerialPortException: Port name - COM4; Method name - setEventsMask(); Exception type - Can't set mask.
	at processing.app.Serial.dispose(Serial.java:144)
	at processing.app.SerialMonitor.close(SerialMonitor.java:108)
	at processing.app.AbstractMonitor.suspend(AbstractMonitor.java:96)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2182)
	at java.lang.Thread.run(Thread.java:745)
Caused by: jssc.SerialPortException: Port name - COM4; Method name - setEventsMask(); Exception type - Can't set mask.
	at jssc.SerialPort.setEventsMask(SerialPort.java:279)
	at jssc.SerialPort.removeEventListener(SerialPort.java:1064)
	at jssc.SerialPort.closePort(SerialPort.java:1090)
	at processing.app.Serial.dispose(Serial.java:141)
	... 4 more
Board at COM4 is not available

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

spacehuhn, 2017-02-06:
Click upload again. If this doesn't work close the serial monitor and open it again after the upload is finished.

nutterdk, 2017-02-06:
Ive clicked upload again, got the same message. So i closed it and uploaded again which i got these error messages
error: Failed to open COM4
error: espcomm_open failed
error: espcomm_open failed

But now when trying to open Serial monitor it tells me "Board at COM4 is not available

nutterdk, 2017-02-06:
Dude my bad, I completely forgot I had unplugged the board :D its uploading now with no errors

nutterdk, 2017-02-06:
Alright so the deauther is uploaded but when scanning, still no networks are showing :/

spacehuhn, 2017-02-06:
Ok this is strange! Because both, the example and my code, use to same functions to scan for APs.
Hm.. do you see any activity in the serial monitor while scanning with the deauther?

nutterdk, 2017-02-06:
This is all i get the monitor
starting...
SSID: pwned
Password: deauther

spacehuhn, 2017-02-06:
Ok well then I have no idea... :(
I never had this problem, I can't reproduce it and it really doesn't make sense! If the example code is working, the deauther code should be too - it's the same!

If you get it running somehow, let me know please!

andrewheselton, 2017-02-06:
Hi Spacehuhn
Great video by the the way. I'm getting the same as nutterdk :( open the example one and serial display 16 access point around where I live but when recompile and upload the deauther shows nothing, very strange.

nutterdk, 2017-02-06:
I'm thinking that I am just going to reset my board by holding down the reset button and reupload/test it and if that doesnt work then I have no idea either

nutterdk, 2017-02-06:
Yeah still nothing

andrewheselton, 2017-02-06:
I'm using one of these, they should be ok right ?

andrewheselton, 2017-02-06:
in the readme it says to go here packages > esp8266 > hardware > esp8266 > 2.0.0 > tools > sdk > include
I have version 2.3 in the esp8266 directory.

nutterdk, 2017-02-06:
You need version 2.0

nutterdk, 2017-02-06:
Yours probably isnt working because you have the wrong sdk version

andrewheselton, 2017-02-06:
Hi nutterdk, do you have a link for 2.0.0 ?

nutterdk, 2017-02-06:
Just delete the one you have now and go follow the tutorial again, it shows you how to do it. It says to select 2.0.0 from the drop down menu after searching for esp8266

andrewheselton, 2017-02-06:
ah yes of course, thanksz i'll try that now.

nutterdk, 2017-02-06:
Tell us if you get it working please :D

andrewheselton, 2017-02-06:
will do having trouble making it show now.

andrewheselton, 2017-02-06:
I can't see the library in the library manager :(

spacehuhn, 2017-02-07:
Just to make it clear, it works now right?
It was a browser problem with IE (?): https://github.com/spacehuhn/esp8266_deauther/issues/6

andrewheselton, 2017-02-07:
Hi StefanThat's correct it was a browser problem probably something to do with the html format.  Many thanks for writing and spending the time with this.ThanksAndy

kpjVideo, 2017-02-16:
Strangely enough I seem to be facing the same issue.

I've got the 2.0 SDK installed, and all the network scans from the example seem to work fine.

I've tried Google Chrome and Safari browser on my iPhone and so far neither of them find any networks are are stuck at "scanning..."

Staple-nutz, 2017-02-16:
I can confirm that I am having the same issue.

After the first compile and upload my Android phone, Windows laptop with Chrome & Windows tablet with IE have all connected to my ESP-12F with Deauther and have all scanned for AP's sucessfully. That was last night.

Today when I have plugged the ESP-12F back into power, I no longer get any APs when scanning. I had not changed anything on the ESP-12F. no re-upload, the same power supply, everything is the same as it was yesterday. Hence I really don't understand what is happening. I have since tried re-uploading the sketch, but am seeing the same result.

I will be trying another ESP board to see if I can get it working on that. Other then this I have no further ideas.

kpjVideo, 2017-02-16:
> Today when I have plugged the ESP-12F back into power, I no longer get any APs when scanning.

Have you tried other browsers for your tablet device? If I had to guess it's something with how the POST information is sent when the buttons are pressed.. Really no idea though.

Staple-nutz, 2017-02-16:
> Have you tried other browsers for your tablet device?

Absolutely
On the Android I have tried the standard browser that comes with the phone as well as Chrome.
On the Laptop I have tried Chrome & MS Edge.
On the Tablet I have tried Chrome & IE 11.

All of which worked yesterday, non do today.

Since my last post I have recompiled and uploaded to another ESP-12F board and I'm still having the same problem on all devices and browsers. 

This is making me think that perhaps some temporary files being stored on the devices are creating an issue after the first use. I'm coming to this conclusion through the process of eliminating...

The ESP HW as an issue, as I've used 2 boards.
The issue started on an ESP that had only ever been flashed one time which at first worked but later didn't.
All attempts to recompile and upload have been the original un-touched sketch
3 Client devices have been used with various browser software. All of witch worked the first time and have later not worked.

andrbmgi, 2017-02-18:
I got similar symptoms as Staple-nutz, it worked and then stopped. A quick debug output to serial in "APScan::start()" reveals that the APs are indeed found correctly. The problem is related to how the data is being sent back to the browser as kpjVideo said. I don't fully understand how this part works though.

spacehuhn, 2017-02-19:
Can you confirm that it works now with the updated files (for all your browsers and devices)?

kpjVideo, 2017-02-19:
I've redownloaded and re-uploaded the new code to the ESP-12E but I still can't seem to get past the "scanning" in both Google Chrome and Safari in my phone browser :/

spacehuhn, 2017-02-20:
I just updated a few things and increased the result list to 80, maybe this will help.

Staple-nutz, 2017-02-20:
> There is a tiny bug in "APScan::getResults()" in APScan.cpp. If you got more than maxResults APs an extra comma gets added to the json which throws off the parser.

>Fixed`` line 73: if ((i!=results-1) && (i!=maxResults-1)) json += ",";

>The timeout is also too short and will fire every time for me, with the fix above a manual reload will do.

After making this change I'm now getting scan results again. I thought nothing in my environment had changed hence some serious head scratching. But it is entirely possible a few more AP's could have been picked up when running at a different time or the ESP orientated differently. Which triggered this bug.

labraloutre, 2017-02-20:
Hi 
I'm having quite the same problem : 
~3 seconds after running scan, I get a timeout error message (the one contained in functions.js)
I've got this behaviour on every device I own (Android phone / Linux Laptop / Win10 Laptop) and with various browsers FF / Chrome / Edge / Chromium.
I've tried to power the ESP from computer and various USB adapters (including 2Amps).

The Wifi Scanner example works well.

I managed to catch the following trace from the serial interface with CuteCom.

Hope this helps and thanks !

```
starting...
SSID: pwned
Password: deauther

Exception (28):
epc1=0x4000df60 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff5330 end: 3fff5690 offset: 01a0

>>>stack>>>
3fff54d0:  4010684c 4010683a 3fff5590 4020e53a  
3fff54e0:  00000af3 00000001 3fff5590 4020e592  
3fff54f0:  00000b03 00000001 3fff5590 4020e95d  
3fff5500:  00000013 3fff08e4 3fff5590 3fff0898  
3fff5510:  00000013 3fff08e4 3fff5590 4020e9bd  
3fff5520:  3ffe8570 3fff08e4 3fff5590 4020e4ec  
3fff5530:  3ffe8a20 00000000 3fff5590 40206e20  
3fff5540:  3fff9000 0000001f 00000011 3fff8ee8  
3fff5550:  0000000f 00000005 3fff5590 4020e5c2  
3fff5560:  3fff0f7b 3fff17b7 3fff1993 00000013  
3fff5570:  3fff2ca8 3fff23e4 3fff5640 00000001  
3fff5580:  00000001 00000001 3fff85a8 402086ae  
3fff5590:  00000000 00000aff 00000af2 4020e5c2  
3fff55a0:  4010684c 3fff85a8 3fff55d0 4020d742  
3fff55b0:  0000000a 401011f3 3fff85a8 4020a8e2  
3fff55c0:  4020a8ec 3fff85a8 3fff55f0 4020a91e  
3fff55d0:  3fff9000 0000001f 00000013 4020e6e8  
3fff55e0:  4020a8ec 3fff85a8 3fff23e4 4020aad9  
3fff55f0:  3fff8fd0 0000001f 00000013 40209b7f  
3fff5600:  3fff2420 401011f3 3fff4670 00000001  
3fff5610:  00000000 0000000a 4020c768 3fff4670  
3fff5620:  402012e6 00000001 00000001 3fff4668  
3fff5630:  3fffdc20 000003e8 3fff23e4 4020abd4  
3fff5640:  3ffe88f0 00000000 000003e8 000092c9  
3fff5650:  3fff2404 3fff90c0 00000016 40101941  
3fff5660:  4020c789 00000000 3fff4661 40208fe1  
3fff5670:  00000000 00000000 3fff4661 4020c7b1  
3fff5680:  00000000 00000000 3fff4670 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
EEPROM cleared

starting...
SSID: pwned
Password: deauther

```

spacehuhn, 2017-02-21:
Thanks @labraloutre for the detailed information. Seems like the ESP crashes after scanning for some reason. I will add more debug infos to the serial monitor with the next update! until then could the others please check if they got the same error?

labraloutre, 2017-02-21:
Hi
I've some informations to add.
I just tried again without changing anything except I'm at work, well it had worked for some time quite well.

I'm wondering whether it Could be something related to the networks surrounding an area, perhaps a SSID name with special characters or accents or too long ?...

JoelBrun, 2017-02-21:
I had the same issue and the same error at my workplace. However at the trainstation 30 mins later it worked again...

spacehuhn, 2017-02-21:
I think either there is an error in the code which lets the ESP8266 crash or there is something wrong with the returned JSON string.
hm... :/

JoelBrun, 2017-02-21:
probably something about the JSON, since it worked at a different place... I can try to scan the APs with the `ESP8266WiFi WiFi Scan`. Maybe there are some chars in it that make it crash.

DushyantPUNK, 2017-02-25:
what is the solution for scanning in mobile devices?

whyisthisusernamealreadytaken, 2017-02-26:
@DushyantPUNK it has nothing to do if you scan it via mobile or pc. There is a bug in the code that probably is caused by an attribute (ssid, mac, etc) of a nearby network.

spacehuhn, 2017-02-26:
I get the same error when an AP has a quotes (") in its SSID because then the generated JSON string is invalid. I'll fix that in the next update!

But I'm sure we are all talking about different errors, just with the same results.

- Browser incompatibility is one thing, you can debug using the developer options and see if the javascript gives an error or something else.
- The other thing is when the ESP8266 crashes. @whyisthisusernamealreadytaken you said it doesn't get to the `getResults()` function, so does it crash before that?

You guys need to debug more! I can't help you without knowing whats wrong. I can't reproduce that error and I've tested it a lot now.

whyisthisusernamealreadytaken, 2017-02-26:
@spacehuhn I tried to add more debug information to your code but because I don't have a lot of knowledge in C I don't really understand the whole program. If you could add more debug options I could give you more information.

tobozo, 2017-02-26:
This is exactly how I found myself trying to implement a serial substitute to the webserver thing #37

spacehuhn, 2017-02-26:
@whyisthisusernamealreadytaken have you tried out the testing branch? https://github.com/spacehuhn/esp8266_deauther/tree/testing I added a lot of debug infos. You just need to set `debug = true` in the esp8266_deauther.ino file. Maybe this helps a bit.

And thank you for your help so far! I hope we can find the error soon :)

labraloutre, 2017-02-26:
Hi @spacehuhn 
I just tried with the testing branch, and the symtoms are the same (timeout and no data on the client web page)
Strangely, I see a correct log in the serial monitor
Here's a link to my log after a fresh reboot

http://pastebin.com/hpfSQNEE

Thanks for your help

spacehuhn, 2017-02-26:
Thanks!
Can you repeat that and paste the log again? I just updated a few things and after the scan it should now print out the json string to the serial monitor.

labraloutre, 2017-02-26:
Here it is
This time, esp crashed. 

http://pastebin.com/Xi7t2hrF

Hope this helps

tobozo, 2017-02-26:
I can reproduce the problem too on the testing branch.

I've added Serial.print statements to the json loop in APScan::getResults() and it seems to crash randomly, sometimes in the middle of serial.print()ing a valid word, doest it mean it's a not a problem with the getters?

Speculation: maybe dealing with HTTP requests while the in scan mode is what makes it crash ?

spacehuhn, 2017-02-26:
`Exception (28)` means `LoadProhibitedCause`:
https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md

> A load referenced a page mapped with an attribute that does not permit loads

I got the same error before when I made a mistake and the generated string was to big and let it crash when printing or sending it.
@labraloutre I see in your log that you have a lot of WiFi networks nearby, maybe you can try to reduce the list size. Go to APScan.h and set `maxResults = 10` (3rd line).

[34 more replies]

closed 2017-03-13 by spacehuhn