SpacehuhnTech/esp8266_deauther #6: How do you reset your board?

nutterdk, 2017-02-06:
Obviously being new to this. How do you factory reset the board so I can start from scratch?

andrewheselton, 2017-02-06:
I got it working :)
I was accessing the esp from my ipad and opening webpage in chrome and not working.  I used my windows 7 laptop and connected and got a shed load of access points showing, I managed to find the library it was in boards not libraries (I'm tired)

andrewheselton, 2017-02-06:
I'm using one of these http://www.ebay.co.uk/itm/152328886050?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

nutterdk, 2017-02-06:
I was using my ipad aswell? Maybe that is it. I shall try my computer
Im using this http://www.ebay.co.uk/itm/NEW-NodeMcu-DevKit-ESP8266-12E-4MB-WIFI-CP2102-USB-Internet-of-Things-IOT-/231983263238?hash=item360347ae06:g:CVkAAOSwnipWU62m&clk_rvr_id=1162855750654&rmvSB=true

andrewheselton, 2017-02-06:
I just tried it on my old android phone and it works ok, its something to do with the ipad/iphone wifi I think

nutterdk, 2017-02-06:
Just just connected it to my pc, and tried scanning for networks using internet exploring. Still nothing

andrewheselton, 2017-02-06:
Mine are the same as yours.  Just tried ie and that does not work but chrome does.

andrewheselton, 2017-02-06:
Firefox under windows 7 works too. Internet explorer is no go.

spacehuhn, 2017-02-07:
Please guys..
Don't spam this much. I just wake up with over 36 new mails!
Don't talk off topic and don't open up useless issues which aren't ones. There is a lot of information online for this board, there are 2 official forums online and you can get help there. This isn't a forum.
When you have problems with the basics, learn Arduino first and start with easier projects. There are so much examples, tutorials and information online.

No offense, I just don't have the time to help out everyone with basic problems like how to reset the board.

closed 2017-02-07 by spacehuhn