SpacehuhnTech/esp8266_deauther #9: Only 20pkts per second in deauth mode

doge27, 2017-02-09:
I'm getting only 20pkts per second in deauth mode, so clients are not getting disconnected. Other functions are working normally

spacehuhn, 2017-02-09:
It's working, the deauth attack is limited to 20pkts/s. 
For more information read the FAQ.

closed 2017-02-09 by spacehuhn

doge27, 2017-02-09:
Clients don't get disconnected. What to do?

doge27, 2017-02-09:
Is it possible to increase that value?

spacehuhn, 2017-02-10:
From the FAQ:

**Deauth attack won't work**
[...]
If it can send packets but your target don't loose its connection then the WiFi router uses 802.11w and it's protected against such attacks or they communicate via 5 GHz WiFi, which the ESP8266 doesn't support.

doge27, 2017-02-10:
How to increase deauth packets? Where to change the value?

sarimnaveed2, 2017-11-10:
Go-to the settings where you change your SSID and Password, go to the Bottom of the page and you'll see the Packets . You can change the value from there.