SpacehuhnTech/esp8266_deauther #11: 'wifi_send_pkt_freedom' was not declared in this scope

jdh024, 2017-02-10:
I have followed the directions exactly but when i go to compile the script it gives the following error.

Arduino: 1.8.1 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

sketch\Attack.cpp: In member function 'bool Attack::send(uint8_t*, int)':

Attack.cpp:41: error: 'wifi_send_pkt_freedom' was not declared in this scope

   if(wifi_send_pkt_freedom(buf, len, 0) == -1){

                                       ^

sketch\Attack.cpp: In member function 'void Attack::run()':

Attack.cpp:179: error: 'wifi_send_pkt_freedom' was not declared in this scope

       if(wifi_send_pkt_freedom(deauthPacket, 26, 0) == -1){/*

                                                   ^

Attack.cpp:190: error: 'wifi_send_pkt_freedom' was not declared in this scope

       if(wifi_send_pkt_freedom(deauthPacket, 26, 0) == -1){/*

                                                   ^

Attack.cpp:221: error: 'wifi_send_pkt_freedom' was not declared in this scope

         if(wifi_send_pkt_freedom(packet, packetSize, 0) == -1){/*

                                                       ^

exit status 1
'wifi_send_pkt_freedom' was not declared in this scope

doge27, 2017-02-10:
Did you modified user_interface.h?

spacehuhn, 2017-02-10:
Have you editited the user_interface.h correctly?
This error only appears if you don't add the needed code lines at the end of the file.

jdh024, 2017-02-10:
I have edited the user_interface.h file with

jdh024, 2017-02-10:
I have edited the user_interface.h file with the 4 lines of script given. 
Also note I updated the board manager with latest version and now it works flawlessly.

closed 2017-02-10 by spacehuhn

doge27, 2017-02-11:
Deauthentication works only on board version 2.0.0. Don't upgrade it.

red0072, 2017-10-31:
this is what i get

 if (wifi_send_pkt_freedom(packet, packetSize, 0) == -1) {

and this

Arduino: 1.8.4 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

sketch\Attack.cpp: In member function 'bool Attack::send()':

Attack.cpp:109: error: 'wifi_send_pkt_freedom' was not declared in this scope

   if (wifi_send_pkt_freedom(packet, packetSize, 0) == -1) {

                                                  ^

exit status 1
'wifi_send_pkt_freedom' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tobozo, 2017-10-31:
have faith in the ReadMe :page_with_curl:

red0072, 2017-10-31:
what does this mean

Arduino: 1.8.4 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

C:\Users\S527321\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp:564:8: error: prototype for 'int8_t ESP8266WiFiClass::scanNetworks(bool)' does not match any in class 'ESP8266WiFiClass'

 int8_t ESP8266WiFiClass::scanNetworks(bool async)

        ^

In file included from C:\Users\S527321\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp:22:0:

C:\Users\S527321\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src\ESP8266WiFi.h:243:12: error: candidate is: int8_t ESP8266WiFiClass::scanNetworks(bool, bool)

     int8_t scanNetworks(bool async = false, bool show_hidden = false);

            ^

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tobozo, 2017-10-31:
It means you're getting closer but still have missed a couple of steps in the ReadMe

saxfraga, 2017-11-04:
I just had the same problem and I was able to troubleshoot it, here is how I was able to fix it:
After you update "user_interface.h" you must add it to the sketch
On Arduino IDE:
Sketch - add file- sdk_fix - user_interface.h (add file to the sketch and upload) Problem solved.
I hope this helps

danilopiona, 2023-09-18:
> I just had the same problem and I was able to troubleshoot it, here is how I was able to fix it: After you update "user_interface.h" you must add it to the sketch On Arduino IDE: Sketch - add file- sdk_fix - user_interface.h (add file to the sketch and upload) Problem solved. I hope this helps

that solved my problem. thanx a lot. Fraternal hug