SpacehuhnTech/esp8266_deauther #12: Fatal error

esp8266error, 2017-02-10:
Whenever I go to upload this file it gives me a long error message

 Arduino: 1.8.1 (Windows 10), Board: "Olimex MOD-WIFI-ESP8266(-DEV), 80 MHz, Serial, 115200"

Build options changed, rebuilding all
C:\Users\*************\AppData\Local\Temp\Temp1_esp8266_deauther-master.zip\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:11:18: fatal error: data.h: No such file or directory

 #include "data.h"

                  ^

compilation terminated.

exit status 1
Error compiling for board Olimex MOD-WIFI-ESP8266(-DEV).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Any idea how to fix? I tried all the models and they all did not work. New Version Node MCU LUA WiFi Internet ESP8266 Development Board Module is the version I got. I believe it is the 512 kb one.

doge27, 2017-02-10:
Choose NodeMCU 1.0 as board and set CPU frequency as 80mhz, flash size as 4M, upload speed as 115200 and select a specific port. It should work

spacehuhn, 2017-02-10:
Yes exactly.
You probably using the wrong SDK because the original doesn't have an Olimex board.
Be sure you downloaded v. 2.0.0 and edited user_interface.h correctly like in the installation steps. 
Then select your NodeMCU and upload.
(by the way all NodeMCUs have 4mb flash).

esp8266error, 2017-02-11:
Arduino: 1.8.1 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (1M SPIFFS)"

C:\Users\********\AppData\Local\Temp\Temp1_esp8266_deauther-master.zip\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:11:18: fatal error: data.h: No such file or directory

 #include "data.h"

                  ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

doge27, 2017-02-11:
Have you added all the necessary libraries?

spacehuhn, 2017-02-11:
Maybe you should unpack the zip to a reasonable place instead of trying to compile the hole software right out of the zip file! 

`Local\Temp\**Temp1_esp8266_deauther-master.zip**\esp8266_deauther-master\`

closed 2017-02-15 by spacehuhn