SpacehuhnTech/esp8266_deauther #13: Include client's signal strength

doge27, 2017-02-10:
Please add clients signal strength as it is in AP.

spacehuhn, 2017-02-10:
This is not possible sorry. There is no way to get the signal strength from user packets with the SDK functions.
http://bbs.espressif.com/viewtopic.php?t=683

closed 2017-02-10 by spacehuhn

doge27, 2017-02-10:
Is it possible to increase deauth packets? Any advantage from increasing it?

spacehuhn, 2017-02-10:
Yes is possible but no won't give you any advantage.

doge27, 2017-02-10:
What I noticed that is the clients which are far from the access point are not getting disconnected. Can you please explain why is that?

spacehuhn, 2017-02-10:
Because they probably don't get the deauth packets. This is a tiny >$2 chip you can't assume it has the same transmition power than a computers Wi-Fi card.

doge27, 2017-02-10:
But I have found that transmit power of esp8266 is set to low to consume minimum current. Is it possible to increase transmit power using setOutputPower?

spacehuhn, 2017-02-10:
just try it out then.

doge27, 2017-02-10:
How?

doge27, 2017-02-10:
Where to define it?

tpmodding, 2017-02-10:
gibt schon penetrante leute

doge27, 2017-02-11:
After setting max TX power, the signal strength increased from -73 to -65