SpacehuhnTech/esp8266_deauther #14: Trouble with NodeMCU 1.0

djcfg, 2017-02-10:
G:\1\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino: In function 'void setup()':

esp8266_deauther:71: error: 'load404' was not declared in this scope

   server.onNotFound(load404);

                     ^

esp8266_deauther:73: error: 'loadIndex' was not declared in this scope

   server.on("/", loadIndex);

                  ^

esp8266_deauther:75: error: 'loadClients' was not declared in this scope

   server.on("/clients.html", loadClients);

                              ^

esp8266_deauther:76: error: 'loadAttack' was not declared in this scope

   server.on("/attack.html", loadAttack);

                             ^

esp8266_deauther:77: error: 'loadFunctionsJS' was not declared in this scope

   server.on("/functions.js", loadFunctionsJS);

                              ^

esp8266_deauther:80: error: 'loadStyle' was not declared in this scope

   server.on ("/style.css", loadStyle);

                            ^

esp8266_deauther:83: error: 'sendAPResults' was not declared in this scope

   server.on("/APScanResults.json", sendAPResults);

                                    ^

esp8266_deauther:84: error: 'startAPScan' was not declared in this scope

   server.on("/APScan.json", startAPScan);

                             ^

esp8266_deauther:85: error: 'selectAP' was not declared in this scope

   server.on("/APSelect.json", selectAP);

                               ^

esp8266_deauther:86: error: 'startClientScan' was not declared in this scope

   server.on("/ClientScan.json", startClientScan);

                                 ^

esp8266_deauther:87: error: 'sendClientResults' was not declared in this scope

   server.on("/ClientScanResults.json", sendClientResults);

                                        ^

esp8266_deauther:88: error: 'selectClient' was not declared in this scope

   server.on("/clientSelect.json", selectClient);

                                   ^

esp8266_deauther:89: error: 'setClientName' was not declared in this scope

   server.on("/setName.json", setClientName);

                              ^

esp8266_deauther:90: error: 'sendAttackInfo' was not declared in this scope

   server.on("/attackInfo.json", sendAttackInfo);

                                 ^

esp8266_deauther:91: error: 'startAttack' was not declared in this scope

   server.on("/attackStart.json", startAttack);

                                  ^

exit status 1
'load404' was not declared in this scope

Any idea?

tpmodding, 2017-02-10:
did you choose nodemcu in arduino ide?

djcfg, 2017-02-10:
Yep? and version in board manager is 2.0.0

djcfg, 2017-02-10:
Problem SOLVED.

load404, loadIndex, loadClients etc... it's a functions. they are not visible, if they are after 'void setup()'

I move it before 'void (setup)' and it works! )))

closed 2017-02-10 by djcfg