SpacehuhnTech/esp8266_deauther #15: Error compileing for board NodeMCU 1.0

76u7h, 2017-02-11:
i keep getting this error when i try to compile the script and i know i the board is set right in tools.

 
Arduino: 1.8.1 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
In file included from C:\Users\Jaye\Desktop\esp8266_deauther\esp8266_deauther.ino:8:0:

C:\Users\Jaye\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:451:2: error: invalid preprocessing directive #typedef

 #typedef void (*freedom_outside_cb_t)(uint8 status);

  ^

C:\Users\Jaye\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:452:2: error: invalid preprocessing directive #int

 #int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb);

  ^

C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:453:2: error: invalid preprocessing directive #void

 #void wifi_unregister_send_pkt_freedom_cb(void);

  ^

C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:454:2: error: invalid preprocessing directive #int

 #int wifi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);

  ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

doge27, 2017-02-11:
Make sure you are using esptool 2.0.0. Edit usee_interface.h correctly. Select the correct port

spacehuhn, 2017-02-11:
The errors tell you that something is wrong with the user_interface.h.

closed 2017-02-21 by spacehuhn