SpacehuhnTech/esp8266_deauther #20: Upload issue

kamiller702, 2017-02-14:
When i compile i get the following:

Sketch uses 518059 bytes (49%) of program storage space. Maximum is 1044464 bytes.
Global variables use 43640 bytes (53%) of dynamic memory, leaving 38280 bytes for local variables. Maximum is 81920 bytes.

Looks good to me.

When i try to upload it though esp8266_deauther arduino file to esp8266 i get the following error

error: Failed to open COM1
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

The tools > port  is grayed out. Tried multiple usb cables and usb ports. Thank you in advance for any help. 
I have no clue what im doing. Just trying to learn. 

Cheers,
Kevin

tpmodding, 2017-02-14:
is it a nodemcu or a esp8266 single board?

kamiller702, 2017-02-14:
https://www.amazon.com/HiLetgo-Version-NodeMCU-Internet-Development/dp/B010O1G1ES/ref=sr_1_3?ie=UTF8&qid=1487101382&sr=8-3&keywords=esp8266

spacehuhn, 2017-02-15:
The ESP upload tool can't communicate with the chip, make sure the right port is selected!
You can also try out different USB ports and cables.
If this doesn't solve it you may have to install USB drivers.
Which drivers you need depends on the board, most boards use a cp2102, cp2104 or ch340.

hevnsnt, 2017-02-16:
Also, make sure that you are rebooting your board into programming mode (this is most likely the issue)

closed 2017-02-21 by spacehuhn

kamiller702, 2017-02-21:
Arduino: 1.8.1 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Sketch uses 522430 bytes (50%) of program storage space. Maximum is 1044464 bytes.
Global variables use 46568 bytes (56%) of dynamic memory, leaving 35352 bytes for local variables. Maximum is 81920 bytes.
error: Failed to open COM1
error: espcomm_open failed
error: espcomm_open failed

kamiller702, 2017-02-21:
downloading usb driver from here 

https://www.pololu.com/docs/0j7/all

kamiller702, 2017-02-21:
"Port" is still greyed out under tools and will not upload to board