SpacehuhnTech/esp8266_deauther #21: Odd issues

BusesCanFly, 2017-02-15:
I'm getting an error when trying to upload to a nodemcu. I am brand-new to the nodemcu, and esp, which means there is a lot of room for error. Help and guidance would be much appreciated!

java.io.IOException: Cannot run program "python.exe": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:206)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:707)
	at processing.app.SketchController.exportApplet(SketchController.java:680)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
An error occurred while uploading the sketch

spacehuhn, 2017-02-15:
Java isn't able to find python.exe. Check if you have python installed correctly.

BusesCanFly, 2017-02-18:
Ah, I should've seen that coming. I'm now running into this error, and my experience with large arduino programs, and the esp8266 is brand-new, so I apologize in advance for potentially dumb questions.
Does anyone understand why this is happening?

Traceback (most recent call last):
  File "C:\Users\_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/espota.py", line 302, in <module>
    sys.exit(main(sys.argv))
  File "C:\Users\_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/espota.py", line 297, in main
    return serve(options.esp_ip, options.esp_port, options.auth, options.image, command)
  File "C:\Users\_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/espota.py", line 91, in serve
    sent = sock2.sendto(message, remote_address)
TypeError: a bytes-like object is required, not 'str'
TypeError: a bytes-like object is required, not 'str'

andrbmgi, 2017-02-18:
Looks like you set it to over the air (ota) upload. In Arduino under Tools, set it to "Upload Using: serial".

closed 2017-02-21 by spacehuhn

itsashishupadhyay, 2018-04-10:
helped by --> https://github.com/per1234?tab=repositories
You need to add the folder that contains python.exe to your path:

    Determine the path to the folder that contains python.exe. You can use Windows File Explorer for this.
    In the search bar, type "System (Control Panel)"
    Click on the matching search result
    Click the "Advanced system settings" link.
    Click the "Environment Variables" button.
    In the section System Variables, find the PATH environment variable and select it.
    Click the "Edit" button.
    Click on the first blank line
    Type the path to the folder that contains python.exe.
    Click "OK"
    Click "OK"
    Click "OK"

bigFin, 2018-04-23:
@itsashishupadhyay This did not work for me.

itsashishupadhyay, 2018-04-23:
@bigFin try restarting the IDE, then uploading via USB and then uploading via OTA ( Do recheck you have added the path to python.exe in Environment Variables (Search Advanced system settings in taskbar---->Environment Variables---------->Path (click edit) add folder path containing python.exe )  )

bigFin, 2018-04-25:
@itsashishupadhyay manually adding the variable did not work. I reinstalled Python and selected the option to install the environmental variable. This worked.