SpacehuhnTech/esp8266_deauther #23: Access Point Not Starting

5yn, 2017-02-15:
Hello there,
I've uploaded the code to my device (NodeMCU ESP-12E).
I see the network appear on both my laptop and my android. But I'm only able to join the network from my phone, and even then I'm not able to access the webpage. It just loads indefinitely.
On my laptop, I'm unable to connect to the network at all.

Upon checking the serial monitor, I see a bunch of random characters (that don't seem to like being copied and pasted) and the following text.

starting...
SSID: pwned
Password: deauther

I've left it running overnight, but it doesn't change from this state.
I've attempted resetting and re-flashing the device quite a few times, but I have the same result each time.

Any help would be fantastic and much appreciated!
Many thanks!

Splodgie, 2017-02-19:
Hi Did you get it sorted?. I had many problems getting it to work but got there in the end.

5yn, 2017-02-19:
Hello! I haven't managed to figure it out yet. I'm still receiving the same errors as before. 
What did you do in order to make it work? 
Thanks for the reply!

Hammad1029, 2017-02-19:
forget network you current and main wifi; a similar problem happened with me and it was because the device would automatically connect to my original wifi

Splodgie, 2017-02-19:
On a Android phone it will only works with FireFox and Chrome, all the other browsers I tested would not work with the program. (tested about 10)
You have to forget all the networks in your WiFi list, (so they will not auto connect) the only one that should be set to connect is for this program ie: pwned / deauthert 
You have to do this because when you scan for Clients (on the AP you selected), it drops the connection to PWNET after scanning and has to reconnect again to PWNET network.
(if there is any password set for you own network it would auto connect to that network instead of PWNET again and the program will fail).
As for you laptop it could something to do with your Virus Checker/ Firewall etc.

5yn, 2017-02-19:
Thanks for the help guys!
It worked once I made my phone forget my home WiFi network and disabled my mobile data. 
It's working like a charm now! 
Thanks again!

closed 2017-02-19 by 5yn

spacehuhn, 2017-02-19:
Great!
I also recommend you to download the current version and upload it. There was a bug which has led to some browser incompatibilities. https://github.com/spacehuhn/esp8266_deauther/issues/5

5yn, 2017-02-19:
Awesome, I'll do that right away! 
Thanks!

Splodgie, 2017-02-19:
Gave the new version a quick test but can only get it to work with FFox and Chrome,  what browser do you recommend?.

Splodgie, 2017-02-19:
Hi again, EEPROM.h is still missing.