SpacehuhnTech/esp8266_deauther #28: Can't change the ssid and the password

Hammad1029, 2017-02-19:
I've edited the ssid and the password char value in the sketch multiple times and also uploaded them multiple times. But the the ssid wont change from "pwned" nor would the password from "deauther"

spacehuhn, 2017-02-19:
Have you tried to reset your board?

Splodgie, 2017-02-19:
Hi, I know nothing about programming, total zip.   This took me a good hour to work out.  
The serial monitor was showing the changed SSID & password put it would not change on the WiFi network.
The new Password must have a Min of 8 letters, ie: abcdergh  then the SIDD and Password will change.
Dave

Hammad1029, 2017-02-19:
splodgie's fix works apparently for some reason, thanks!. sh i hope you look at this for a fix

spacehuhn, 2017-02-19:
Great!
WPA(2) encryption needs at least an 8 character long password, this can't be be changed or "fixed".

closed 2017-02-19 by spacehuhn