SpacehuhnTech/esp8266_deauther #29: [Request] Multiple AP (Please read)

VoidTyphoon, 2017-02-19:
Hi,

I have multiple AP's in my network all with the same SSID and setup so that my devices automatically switch AP's when moving around the house, I found yesterday while trying to stop my phone from connecting it defaulted to another AP in the house, Would it be possible to implement a feature to Block for ALL the Wireless Access Points with the same SSID?

I'm aware that you won't implement the Multiple SSID blocking and I totally agree with the reasoning, however, could this please be considered? also the support in this mode for scanning all the devices on all the hotspots with the same SSID, not just the one?

Thanks -Jack

spacehuhn, 2017-02-19:
OK I understand your problem. I will work something out!

VoidTyphoon, 2017-02-19:
Awesome. Thanks for the quick reply too 👍

spacehuhn, 2017-02-20:
Are your APs running on different channels?

spacehuhn, 2017-02-24:
Implemented here: https://github.com/spacehuhn/esp8266_deauther/tree/testing
please report back if it works with your APs :)

closed 2017-02-27 by spacehuhn