SpacehuhnTech/esp8266_deauther #31: [Feature request] Add custom SSID beacon spam to this repi

nisc3, 2017-02-20:
It would be great if you add custom SSID beacon spam to this repo.

spacehuhn, 2017-02-20:
working on it.

nisc3, 2017-02-20:
Superb!

closed 2017-02-20 by nisc3

xdavidhu, 2017-02-22:
I would be happy to see this feature too!

----------------

My idea would be to add an attack menu for the custom beacon spam, with a textfield there, and the program would generate an array from the textfield, every name separated by the new line ("\n"). And after just go trough the array with a forever loop and send those beacons.

dvdcd, 2018-11-02:
@spacehuhn Is this still planned? It sounds like a fantastic feature.

tobozo, 2018-11-02:
[CTHULHU mode FTW](https://github.com/spacehuhn/esp8266_deauther/blob/master/web_interface/ssids.json) :octopus: