SpacehuhnTech/esp8266_deauther #32: Testing DeAuther

Splodgie, 2017-02-20:
Hi All,
I have tested this Program using the following results.
1..I have both the New an Old Versions of this program Running on ESP8266-01 ()modded to 4MB) x2.
2..I have both the New an Old Versions of this program Running on ESP8266-12E x2.
3..I have tried the New an Old Versions of this program Running on WeMOS D1 x1.
So I have 5 ESP's all loaded with the software.

Android Phones used for testing DeAuther.
1..HTC1 M7.
2..HTC Desire X
3..Sony Xperiac1505
4..Samsung_GT-N7000
5.. Xiaocai x9S
6..LG GT540
7..Iphone 3GS (Sorry about this but I did test it).

Laptop Used.
1..Old DellD540
2..eMacchines E625 (Changed processer tp 2gHz).
3..Old Sony Vaio Laptop.

PC's
1..4x Home built PC Ranging from Duel Core up to i5's (all Clocked)
2..Orange PiPC

Router's Tested.
1..Virgin Media SH V3
2..BT HHv3 (Unlocked)
3..NeXX MiNi Router WT1520
4..Netgear WPN824

Results
I Tried all Combinations of setup, Trying to get one device to block another. took about 6Hrs.
None would DeAuther any other in testing.
Beacon Spamming worked all the time. You cannot target the Client you selected in in the Clients list only.
Beacon Spamming Targets all devices connected to the selected AP list you targeted.

The only Browsers that worked for me is FireFox and Chrome (Chrome would auto reconnect after Client scanning) FireFox you have to hit try to connect again yourself.

Dave

spacehuhn, 2017-02-20:
To get you right, the deauth attack won't work with any of your devices?
Also you have some troubles using different browsers -> https://github.com/spacehuhn/esp8266_deauther/issues/5 (just updated minitues ago, please check out).

Splodgie, 2017-02-20:
Correct, I have NOT got it to Deauther/Block any device, Downloaded the latest version trying It now get back to you ASAP with update.
(What Android browser do you want me to test if any)

spacehuhn, 2017-02-20:
Ok. Have you read this(?):

> Deauth attack won't work  
If you see 0 pkts/s on the website you have made a mistake. Check if you have followed the the installation steps correctly and that the right SDK installed, it must be version 2.0.0!
If it can send packets but your target don't loose its connection then the WiFi router uses 802.11w and it's protected against such attacks or they communicate via 5 GHz WiFi, which the ESP8266 doesn't support.

Splodgie, 2017-02-20:
Yes I have read it, Plus everything connected with DeAuther, Versin 2.0.0 has always been used as per your instruction.
It always send about 20pkts when I sent the attack.
Newest version installed, will start testing.
I used old routers to connect to that should not have the firmware update that block this type of attack.
Dave

VoidTyphoon, 2017-02-20:
To also add to this, I have NOT been very successful on deauthing and it's not 100% effective.

Splodgie, 2017-02-20:
VoidTyphoon, What browser have you been using?.

VoidTyphoon, 2017-02-20:
Mainly FireFox, Although i have tried Chrome Mobile still to no avail.
There may be a problem in the code as it was working on an earlier release.
If i get some spare time later i'll fork it and have a play around, if i manage to fix it ill stick a pull request in and we can get it fixed in the main repo :) 

That is unless @spacehuhn manages to patch it before.

spacehuhn, 2017-02-20:
Would be great if you find the problem! 
I have no issues at all and I have tested it with a lot of different ESP8266 boards, networks and devices.

What you could try is using linux tools to make a deauth attack manually - if this works but the ESP8266s attack doesn't you could use wireshark to analyse if there is something wrong with the packets.

VoidTyphoon, 2017-02-20:
You have pretty much read my mind!!
I will setup a wireless network later and pipe it thru wire shark and attack it from my ESP first and then 
Kali as i know that works on my network.

Out of interest however, what wireless protocol are you testing yours on?

Splodgie, 2017-02-20:
Hi, Still the same with the Latest Version no blocking, Still have to use FF & Chrome, No others work for me, (dose not complete the scan).
I plugged in 3 old routers and made sure I could connect to them, then get internet access through them to my main Router, My main router is protected from this kind of attack, but the 3 old router should be defenceless because of there age 3yrs 4yrs 6yrs old, all only 2.4gHz. I used 3 different phones and one laptop for this quick test and none was blocked. The attacks always sends 20pkts.

Q:  You never include EEPROM.h /cpp ? could this be my problem. 
Dave
???????????? Brain working, must turn it OFF.

Splodgie, 2017-02-20:
Hi, Just read VoidTyphoon comment about what wireless protocol I was using, Just tested again with network set to Open, Still the same no blocking.
Dave

VoidTyphoon, 2017-02-20:
@Splodgie In attack.h go to line 64 https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/Attack.h#L64
and change the value from 0x01 to 0x05 and see if that unveils anything, i have successfully got it going on my network now, just waiting on #29 to be implemented now 😄

spacehuhn, 2017-02-20:
Good to hear. If the reason-code is the problem then we can fix this pretty easy. I'm started to rewrite the attack class and enable multiple APs and advanced beacon flooding.

VoidTyphoon, 2017-02-20:
Great! Looking forward to it, maybe an option could be to select a reason from a dropdown menu? 
A great list i found here http://www.aboutcher.co.uk/2012/07/linux-wifi-deauthenticated-reason-codes/

-Jack

Splodgie, 2017-02-20:
VoidTyphoon, Made the change suggested, I get no difference, no blocking and can only use FF & Chrome.
Dave

Splodgie, 2017-02-20:
Hi, I have asked about EEPROM.h not being included in the download twice, as it is not included in the code, so like others I have been placing it in the esp8266_deauther folder with the others that are included, then it would compile OK.
I have now removed EEPROM.h and EEPROM.cpp from the esp8266_deauther folder, and placed 
#include <EEPROM.h> into the code instead. and it compiles OK as it should.
I have now found that the device being Attacked has to be within 100mm of the ESP and it will block it, BUT.
It will do this to a 12 year old (Aprox) TP-Link TP-WR641G Router.
It will not block the same device when it's connected to a BT HHv3 (Unlocked)   NeXX MiNi Router WT1520   Netgear WPN824 all of then are years old. 
Even when the same device to be blocked is touching the ESP. (any of the ESP I have already listed in my first comment) with any Version of the software and with the Line mod recommended by VoidTyphoon.

This seems very strange as I have had a ESP-10 working at a range of 280 Meters using my HTC1 (Line of sight) Switching a light On and Off.  Q: whats killing the ESP TX power.

Splodgie, 2017-02-21:
Hi All, I have done more testing after finding out My Problem was TX power from the ESP.
I found it's the TX power transmitted by the IP Router being Targeted, and the TX Power of the ESP.
The ESP need to have a 55% more radiated TX power than the the IP Router selected, 
ie: ESP 100%   IP Router 45%,  Blocking is achieved 99% of the time.
ie: ESP 100%   IP Router 50%,  Blocking is achieved 5% of the time, intermittently.
ie: ESP 90%   IP Router 50%,  Router wins, No Blocking.
ie: ESP 90%   IP Router 40%,  ESP wins, Blocking.
ie: ESP 80%   IP Router 35%,  ESP wins, Blocking.
(as long as the IP Router can be compromised, more testing needed).
Thanks All for your efforts and help. This old man needs some sleep. 
Dave

Splodgie, 2017-02-21:
Forgot Something,  The Power readings are taken from the targeted device, This you are not able to see if trying to block a unknown target, not under your control. So this you will have to guess correctly for it to work, or just hope for the best.

spacehuhn, 2017-02-21:
The tx power should be irrelevant. Either the client gets the deauth packet (then it should disconnect) or it doesn't recieve it. Like I said you can test this with wireshark and other linux tools.

Splodgie, 2017-02-21:
Hi, thanks for your reply, It will only work for me if the criteria I stated is met, it works on all the phones and routers as long as it's within what I have stated, All routers ,All Phones.
I must have some thing wrong with all my devices, but that hard to believe, or I have missed something along the way, but what I cannot see.
If some one else is following this please do a test and let us know your results.
Many thank for all your hard work and effort. I will give it a rest for now as I am not getting the results you say and are totally lost as to why, (Old Age).
Regards
Dave

closed 2017-02-21 by Splodgie

chinahk, 2017-02-22:
splogie, me folow how you say to install and it work ok or no instal   me use chrome and it work fire fox as well  no mor browser work 
but program not jam not stop workin phone    i reed what you say for test an test you say do
me test same as yoiu say and work all time jam must be mor power signal esp  not router  jam al time wot you test is true me find 
me find same or not work 55+ mor esp ok less not work
thak for all people help chi.

Splodgie, 2017-02-22:
Hi chinahk, thanks for doing your test and letting us know what you found out, looks like it only working the same for you from what you are saying.
Dave
PS: Glad you got it working as I lost your comments on YouTube.

FirstPure, 2017-02-22:
Hi looking for help with this and found the information in this closed section. could not get this to work after installing went ok after some reading just would not boot any phone off. read the information that you found and tested it the much same way but cannot tell how to see the signal.
but its like you say, jammer has to be very close to the phone your attacking for it to work 3/4 or more from the AP then it works just. how you going to find that out when you cannot see the phone your attacking or the AP its using its good if you want to jam your self 'He He'.

Splodgie, 2017-02-23:
Hi FP. Thanks for posting your results, so it's not working as it should for you as well.
 I could not find out why but a tachie M8 said that it's probably down to the hand shaking between the AP and the Client keeping the connection alive, and that it's only blocked when the AP cannot get back to the client because of the blocking being the stronger signal.
Hhe suggested to test another way, by blocking the client fully then moving the client back into range of the AP, the Client just connects to the AP again. must be like he says the stronger AP H/S signal and the Clients blocking weaker signal.

spacehuhn, 2017-02-23:
Maybe the devices reconnect faster (within 1sec)  than they get the next deauth packet. But even then because of the constant reconnecting, the network speed must be super slow.

Splodgie, 2017-02-23:
Hi, I just started testing the USB WiFi Dongles I have with very good results to the attack being successful, Will post more information when I get further.
 I am loading Android Remix OS on the PC to see if it could be something to do with the OS used was well, but I doubt that very much.  
Dave

Splodgie, 2017-02-23:
Hi, The only thing it Blocks good that own is a iPhone 3GS. it will block every time. 
My Laptops are not Blocked no matter how close the ESP is them even if the AP signal is 20%.
USB dongles, I have 5 of them TPlink, Netgear, EDUP, None of these would Block
Two none branded dongles. 1 can be blocked, but if it has more than a 70% AP signal then it's NOT blockable.
The other USB can be blocked when the AP signal drops below 85%.
None of the Android phones are blocked unless the ESP has a 55% stronger signal to the phone.

If some one who is following this, who have a later iPhone/iPad/AirPad give it a test, post your finding.
That's it from me, done all the testing I can  
Dave
(Spacehuhn have you had a look at the random channel not working on the esp8266_beaconSpam)