SpacehuhnTech/esp8266_deauther #33: Error while compiling

DR13, 2017-02-20:
I get this error while compiling:

`Arduino: 1.6.12 (Windows 10), Board:"NodeMCU 1.0 (ESP-12E Module), 160 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\Dioni\Desktop\Projects\Deauth\X\esp8266_deauther.ino:17:20: error: redefinition of 'const char* ssid'

 const static char *ssid = "pwned";

                    ^

X:17: error: 'const char* ssid' previously defined here

 const static char *ssid = "NodeMCU";

                    ^

C:\Users\Dioni\Desktop\Projects\Deauth\X\esp8266_deauther.ino:18:20: error: redefinition of 'const char* password'

 const static char *password = "deauther"; //must have at least 8 characters

                    ^

X:18: error: 'const char* password' previously defined here

 const static char *password = "123456789"; //must have at least 8 characters

                    ^

C:\Users\Dioni\Desktop\Projects\Deauth\X\esp8266_deauther.ino:20:24: error: redefinition of 'ESP8266WebServer server'

 ESP8266WebServer server(80);

                        ^

X:20: error: 'ESP8266WebServer server' previously declared here

 ESP8266WebServer server(80);

                  ^

C:\Users\Dioni\Desktop\Projects\Deauth\X\esp8266_deauther.ino:28:10: error: redefinition of 'NameList nameList'

 NameList nameList;

          ^

X:28: error: 'NameList nameList' previously declared here

 NameList nameList;

          ^

C:\Users\Dioni\Desktop\Projects\Deauth\X\esp8266_deauther.ino:30:8: error: redefinition of 'APScan apScan'

 APScan apScan;

        ^

X:30: error: 'APScan apScan' previously declared here

 APScan apScan;

        ^

C:\Users\Dioni\Desktop\Projects\Deauth\X\esp8266_deauther.ino:31:12: error: redefinition of 'ClientScan clientScan'

 ClientScan clientScan;

            ^

[... 265 lines cut ...]

 void sendAttackInfo(){ server.send ( 200, "text/json", attack.getResults()); }

      ^

C:\Users\Dioni\Desktop\Projects\Deauth\X\esp8266_deauther.ino: In function 'void startAttack()':

C:\Users\Dioni\Desktop\Projects\Deauth\X\esp8266_deauther.ino:160:6: error: redefinition of 'void startAttack()'

 void startAttack(){

      ^

X:160: error: 'void startAttack()' previously defined here

 void startAttack(){

      ^

exit status 1
'const char* ssid' previously defined here

`

How do I fix this?

VoidTyphoon, 2017-02-20:
uninstall all of your ESP library's and follow the instructions again EXACTLY word for word then get back t me ;)

DR13, 2017-02-20:
I will

DR13, 2017-02-20:
Allright, will do.

DR13, 2017-02-20:
It took a few tries but it works just fine now. With projects like these I tend to go way to fast and I end up reading only half of the text.

Thanks!

closed 2017-02-20 by DR13