SpacehuhnTech/esp8266_deauther #34: [Feature request] settings menu in the webinterface

spacehuhn, 2017-02-20:
What could be usefull for a settings menu?
Ideas and suggestions are welcome!

- change SSID & password
- edit namelist (a table with all saved custom client names)
- list of custom beacon SSIDs
- change the reason code of deauth packets -> https://github.com/spacehuhn/esp8266_deauther/issues/32

poslogic, 2017-02-20:
Would it be possible to be able to set the time that a deauth or beacon spam lasts for e.g. 5mins, 10mins?

VoidTyphoon, 2017-02-20:
The ability to change Beacon & Deauth intervals on the go?

spacehuhn, 2017-02-20:
with interval do you mean the packet rate?

VoidTyphoon, 2017-02-20:
yeah, so say you specify a wait of 1500 in the web browser or any other number

nisc3, 2017-02-20:
Setting max TX power on esp8266.
What about using 
void ESP8266WiFiGenericClass::setOutputPower

    if(dBm > 20.5) {
        dBm = 20.5;
    } else if(dBm < 0) {
        dBm = 0;
    }

    uint8_t val = (dBm*4.0f);
    system_phy_set_max_tpw(val);
}
To set esp8266 to run at max TX power?

spacehuhn, 2017-02-20:
changing the tx power over the settings could be an option, but why? It uses the max power by default and I don't see any reason to change that, given the fact that 20.5dbm isn't really much.

JoelBrun, 2017-02-21:
It would be nice if the AP's from the Beacon-Spam had a password protection. On a mac you can see a small lock symbol next to the ssid. Like that one can easily notice the right network.

Also it would be useful if one could "deauth all but" and then select some mac adresses

nisc3, 2017-02-21:
Does SDK allows more than 20.5 dBm?

spacehuhn, 2017-02-21:
> Does SDK allows more than 20.5 dBm?

Well...have you read the code you posted?

`if(dBm > 20.5) {
    dBm = 20.5;
}`

nisc3, 2017-02-21:
Oops sorry, what I meant is, there is no way to increase it because SDK is closed source right?

spacehuhn, 2017-02-21:
Maybe you could 'overclock' the tx power somehow but I don't think anyone did that before :/

nisc3, 2017-02-22:
It is possible to boost TX power but it causes instability to the board because supply of lack of power...

closed 2017-02-27 by spacehuhn