SpacehuhnTech/esp8266_deauther #36: Why SKD 2.0.0

tobozo, 2017-02-21:
Not a real issue here, but I tried to change the maxResults value to 60 and of course ran into problems.

So I was wondering : why does this project absolutely requires the SDK 2.0.0?

spacehuhn, 2017-02-21:
Well you could just try to run this with a newer SDK version...
You will see that it won't send any packets when starting the deauth attack. That's because the SDK normally doesn't allow that.

Your problem with the maxResults has nothing to do with the SDK, but you can download the newest version which has maxResults set to 80 :)

closed 2017-02-21 by spacehuhn

poslogic, 2017-02-21:
The wifi_register_send_pkt_freedom was removed.

http://bbs.espressif.com/viewtopic.php?t=1357