SpacehuhnTech/esp8266_deauther #37: [Feature request] full serial peering

tobozo, 2017-02-21:
Having to constantly switch between browser / wifi properties / reconnect / refresh sucks.

As this was getting on my nerves, I wanted to try a less disruptive approach by using the stable serial console instead of the unstable wifi connection. This is what I've done so far: https://github.com/tobozo/esp8266_deauther_serial

This version talks full JSON, and will only accept orders coming from the hardware serial port:

  - By using a terminal (i.e With Android Termux on a Samsung Galaxy, the smartphone seems to be able to communicate through serial and operates the ESP8266 quite wel)l.
  - By using a webserver to process+relay orders and data between the ESP's USB Serial and a browser

Now I'm not requesting you ditch the web server like I did, I'm just suggesting using the serial as an _alternate_ channel, provided the other end know how to deal with the data.

spacehuhn, 2017-02-21:
Great idea! I never thought about that, but sure with an app you could use it with your phone what most people already do.
Maybe a native Android app could also be an option? Using the same html files as the user interface but communicate via serial instead of WiFi?

tobozo, 2017-02-21:
If I can get node-serialport to compile on Termux (currently giving errors with node-gyp) the native app is not required, and there's a possible codebase that can also be used from Mac/PC/Raspberry.

If unsuccessful I'll try setting up a Phonegap environment, the Cordovaduino plugin seems good enough for that. However, a native application requires a distribution center, not very rogue-ish :)

I'm more attracted to the node solution though, if I can manage to get node-serialport to install, the only requirement is to have the Termux application installed, run the node application, and open a browser.

spacehuhn, 2017-02-22:
But with node-serialport the user needs to run a node.js server, right? Isn't that a bit complicated on let's say an Android phone?

tobozo, 2017-02-22:
It's not really complicated; 
- install Termux on Android
- open Termux; 
- apt install node make python2-dev g++ vim; 
- edit config with vim if necessary
- run server

While I'm not sure Termux exposes the serial port, or what /dev/tty* is linked to the usb otg (or if it's linked at all), I'm still looking for a context where Serial port and nodejs execution are allowed from a single application.

If I dont' find one, I'll go for the Android native app.

tobozo, 2017-02-23:
Got the Phonegap + Cordovaduino plugin to compile using grunt-phonegap and successfully connected to the OTG Serial port while the ESP8266 was plugged in.

I thought no adb debug was possible while the USB is connected and gave up last night; but I just found out adb debug can be done over the network so I'll check it out later tonight.

If this turns out to be working, I'll publish the project source on github, Is "android_deauther" a good name or should it have a complementary name such as "android_esp8266_deauther" ?

spacehuhn, 2017-02-23:
It would be great if you could add the serial interface to the main repo, just so that both serial and WiFi is an option. I will add "WiFi on/off" to the settings later together with some over things.

That way others could also create Apps or other GUIs easily.

tobozo, 2017-02-23:
So Wifi on means Serial off?
I'll start with that and make a PR soon: 

  #define WIFI_UI_ON true

tobozo, 2017-02-23:
#38

spacehuhn, 2017-02-24:
Hey so I ran into some serious conflicts with your PR with my current codebase. I suggest I finish my work and then we work together to implement a good serial interface.

tobozo, 2017-02-24:
yikes I saw the diff mess ... I had to move loop() and setup() at the end of the ino file otherwise my Arduino IDE (Ubuntu) would complain, I'm not sure how this affects other versions of the IDE or why it happens though.

Meantime I'll see how far I can go with Corvodaduino.

spacehuhn, 2017-02-24:
Great! I would love to see an App for this project!

tobozo, 2017-02-25:
Current status on Android app: connecting well with Cordovarduino although I'm experiencing some data loss problems I can only reduce by modifying the esp8266_deauther behaviour (enumerating AP list items in separate JSON messages, 'pinging' the Android by constantly re-sending the AP List).

I'm not sure whether I should implement some CRC check or just manage a timed message queue, or both.

tobozo, 2017-02-26:
I'm finaly getting somewhere with the sketch and the android app binary available here https://github.com/tobozo/esp8266_deauther_serial

I know it sorta breaks the repository so I'll stop here until we discuss the next steps, the android app is only a proof of concept and it comes with requirements that can be annoying to fulfill.

tobozo, 2017-03-03:
I've also started implementing a substitute to the ESP8266WebServer code in order to minimize the impact on the code in the ino file.

it's just a matter of changing this:

    ESP8266WebServer server(80);

by this:

    SerialServer server(115200);

I'll provide the code on request if it proves to be of any interest.

spacehuhn, 2017-03-03:
Great 👍 
Unfortunatly I still haven't had time to look into your version and the App :(
I'm busy debugging the testing branch. If you're able to make a stable version with both WiFi and serial let me know :) No stress though, we have time!
By the way - WiFi and serial could just run both at the same time (with the option to disable WiFi in the code and via a serial command).

tobozo, 2017-03-03:
Here's the [SerialServer library](https://github.com/tobozo/SerialServer) I've used, it needs to be installed in the 2.0.0/libraries/ folder and cannot yet coexist with the ESP8266WebServer but  it works just fine with a copy of the testing branch cloned a few minutes ago.

tobozo, 2017-03-04:
Just [commited](https://github.com/tobozo/SerialServer/commit/109a54d4c5b86a970eb98373224ddd3e418113c2) so both libraries can be loaded without namespace collisions.

The `#include <ESP8266WebServer.h>` does not need to be commented anymore.

Also I haven't removed yet my first attempt at producing a handlers lists (was coded before you replied to the manifest.json request).

You can get this handlers list by calling `server.getManifest()` while using ServerSerial.
I'm not sure I'll keep this unless ESP826WebServer and RequestHandler can be overriden externally to maintain the syntactic substitution between both libraries.

tobozo, 2017-03-05:
Full serial peering for desktop achieved using a nodejs pseudo proxy, [instructions are in the readme](https://github.com/tobozo/node-deauther).

There are still reasons to build an Android app for the esp8266_deautheer but none of those involve using the serial interface because it's too much of a drag.

tobozo, 2017-03-09:
[This project claims to do OTG](http://www.smartarduino.com/view.php?id=95104) but I'm quite sure it's also using Cordovarduino.

XAS-712, 2017-08-25:
Why not use a HC-05 Bluetooth serial port module? Most android phones support Bluetooth.
The device needs not to have the OTG function.
And this module is cheap. About 15 RMB in China.

tobozo, 2017-08-25:
@XAS-712 Sound good for a headless build, I'll order one right away :-)
There's a physical requirement for spatial isolation between the antennas to avoid interferences though.

dglaude, 2017-09-24:
Creazy idea while reading this issue.
Maybe using two ESP8266, one for the "attack" wifi and one for the "control" wifi.
In between, the @tobozo serial interface that replace the WebServer.

tobozo, 2018-03-24:
achieved in the next version, closing this

closed 2018-03-24 by tobozo