SpacehuhnTech/esp8266_deauther #39: ClientScan crash

spacehuhn, 2017-02-24:
I'm working on the implementation of multiple APs and found an issue with the ClientScan class. I changed it so that the scan scans in all selected APs and changes channels if necessary.
It works fine but when I start the scan for the second time I get this in the serial monitor:
`Exception (0):
epc1=0x401063b9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

[...]`

I created a new brach **testing** with my current progress. Would be great when someone could test this too, maybe you'll find the error. But note: that's just for testing purposes on the ClientScan class, the Attack class is not ready yet and will not work properly!

spacehuhn, 2017-02-24:
Nevermind. I found the stupid bug. I forgot to clear the channel array when starting the scan.

closed 2017-02-24 by spacehuhn

tpmodding, 2017-02-24:
do you create a PR? so i can update my nodemcu and test it now :)

spacehuhn, 2017-02-24:
In a few minutes, just finishing my work... the attacks should also work then! (exept of a little bug with the beacon attack, so don't select to much targets).

spacehuhn, 2017-02-24:
Updated to the testing branch: https://github.com/spacehuhn/esp8266_deauther/tree/testing
Thanks for testing 👍

tpmodding, 2017-02-24:
did you remove the beacon spam and random beacon spam in testing branch? will update now :)

spacehuhn, 2017-02-24:
beacon spam is still there, but I removed the random spam. I will add it later with a list so that you can generate or add cutom SSIDs.

tpmodding, 2017-02-24:
clientscan works now without crashing... IF there is no other network, cause the iphone connects to another network haha but works :)