SpacehuhnTech/esp8266_deauther #41: esp8266_deauther .html not able to modify

sounakdeb, 2017-02-24:
How to modify the esp8266_deauther .html files i try but not working.
i'm edit the index.html then upload its not showing in the browser.

spacehuhn, 2017-02-24:
Use the `minifier.html` in html files to minify and then 'byte-ify' your changed code. In data.h you can then replace the byte array and upload.

sounakdeb, 2017-02-25:
thank you it's working now

sounakdeb, 2017-02-25:
deauther only block single wifi network not available multiple wifi network selection options. can you please add multiple wifi selected option.

closed 2017-02-25 by spacehuhn

garustea, 2020-10-23:
@spacehuhn @sounakdeb 

I am trying to perform the procedure for editing html files,

- i installed anglerfish

-I run the script from its location after modifying the files in web_interface

- i've upload the files via sketch data upload from the arduino IDE

I start the esp8266 but the changes don't appear, where am I wrong?

*before uploading the files via IDE I flashed the bin file via ESP8266 FLASHER

garustea, 2020-10-26:
@Bommlon 
> Did you make sure 'indexhtml' was written correctly? Because this error usually means that there's no variable with that name in the code.

yes I checked the code, I also opened the updated file in data / web and I checked that the contents of index.html.gz were written correctly.

What should be the step by step procedure to edit the html? Were you successful?

thanks for the support

spacehuhn, 2020-10-26:
The procedure to is very easy as long as you only modify existing files (without adding, renaming or removing files):
1. Make your changes to the files in the folder [`web_interface`](https://github.com/SpacehuhnTech/esp8266_deauther/tree/v2/web_interface)
2. Run [`utils/webconverter.py`](https://github.com/SpacehuhnTech/esp8266_deauther/tree/v2/utils/web_converter) 
3. Upload the code using Arduino

If you add or remove files, you'll need to edit [`wifi.h`](https://github.com/SpacehuhnTech/esp8266_deauther/blob/v2/esp8266_deauther/wifi.h#L233) as well.

If you make changes and don't see the result, try clearing the browser cache.

garustea, 2020-10-26:
@spacehuhn 
> The procedure to is very easy as long as you only modify existing files (without adding, renaming or removing files):
>
> 1. Make your changes to the files in the folder [...]

Thanks for replying, The procedure you indicated is the one I followed. When I try to upload the .ino file via IDE I get this error:

```
More` than one library found for "EEPROM.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ EEPROM
More than one library found for "ESP8266WiFi.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ ESP8266WiFi
More than one library found for "SSD1306Wire.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ esp8266-oled-ssd1306-4.0.0
More than one library found for "ESP8266WebServer.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ ESP8266WebServer
More than one library found for "ESP8266mDNS.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ ESP8266mDNS
More than one library found for "Hash.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ Hash
More than one library found for "ArduinoJson.h"
Used: C: \ Users \ myname \ Documents \ Arduino \ libraries \ ArduinoJson
Not used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ ArduinoJson
More than one library found for "Wire.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ Wire
More than one library found for "SimpleButton.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ SimpleButton
More than one library found for "DNSServer.h"
Used: C: \ Users \ myname \ AppData \ Local \ Arduino15 \ packages \ deauther \ hardware \ esp8266 \ 2.0.0 \ libraries \ DNSServer
exit status 1
'indexhtml' was not declared in this scope
```

Have no idea what the problem could be?

I want to clarify that to run the python script, I load the files on a usb key that I insert in raspberry with the kali linux operating system, once the script has been executed I reload the files on a usb key and load them from windows10 via arduino IDE, it could depend on it?

Thank you,

garustea, 2020-10-26:
[image]

garustea, 2020-10-29:
Does anyone have any idea what the problem might be?

spacehuhn, 2021-01-14:
This should be fixed, but for reference:
The compiler doesn't like the `-`. Instead of `pt-brlang` you could call it `ptbrlang` or `pt_brlang`.