SpacehuhnTech/esp8266_deauther #42: Testing Branch

spacehuhn, 2017-02-27:
https://github.com/spacehuhn/esp8266_deauther/tree/testing

Changelog:
- added multiple AP selection
- added Settings
- added beacon list attack
- added SSID list
- optimized JSON/AJAX (bugfix for #5 )

I want to merge this as soon as possible but there are probably new bugs in it so please test and report here if everything works :) thanks.

poslogic, 2017-02-27:
Latest Testing Branch - 27/02/17 9am gmt
NodeMCU ESP8266
Android/Chrome

All working great deauthing my Chrombook, Intel Laptop(Ubuntu), Android TV box and Raspberry Pi very successfully at different distances from my AP :-)

New settings working as expected.

Beacon (clone) working as expected.
Will there be another option to use Beacon (spam) that will use the SSID list?

Fantastic work, thank you for all the work you have done on this project :100: :+1:

tobozo, 2017-02-27:
Latest Branch Feb 27th 2017 19:16 (Paris time)
NodeMCU ESP8266 dev kit
Android/Chrome

First scan => working well.
Select an AP => working well
Scan AP => finds nothing (should find three of my devices), disconnects me, and shows me an empty AP list on reconnect.

Second scan => Exception (28), here's the dump (debug enabled): http://pastebin.com/pLmCHTFs

DushyantPUNK, 2017-02-27:
Latest Branch Feb 27th 2017 5:30
NodeMCU ESP8266 dev kit
Symbian/Opera
No Ap are showing after pressing the Scan button. It should show me 10-14 AP.
Working fine on Laptop Chrome Browser.

whyisthisusernamealreadytaken, 2017-02-27:
Latest Branch Feb 27th 2017 08:00 CET
NodeMCU ESP8266 dev kit
Opera (Mobile)

Everything is now working! Good job. However I have a suggestion:
The custom SSID list should have an option of spamming. 
If I put one SSID in the list it would spam the beacon as if I selected an AP with that name.

H1ghTech, 2017-02-27:
Latest Branch Feb 27th 2017 20:00 CET
ESP8266 12 standalone
Chrome + Firefox (Mobile) | Firefox (Desktop)

Everything is working exept: When I select a Ap and try to deauth it it will turn off the Ap of my Esp. After reconnecting andreloading the page it will show 0pkts/s

spacehuhn, 2017-02-27:
@H1ghTech seems like it's crashing, you can set `debug = true` and have a look what it's printing to the serial monitor.

H1ghTech, 2017-02-27:
SSID: TBU
SSID length: 3
password: deauther
password length: 8
deauth reson: 0
attack timeout: 0
attack packet rate: 0
client scan time: 30
attack SSID encrypted: 0

starting...
starting WiFi AP
SSID: TBU
Password: deauther
generating Macs...done
getting attacks JSON...done 
getting attacks JSON...done 
getting AP scan result JSON done
starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID
e0:28:6d:39:15:c0 - 1 - -91 - WPA2 - TBU_ZERO
2c:30:33:ce:1d:ef - 9 - -85 - WPA2 - TBU
scan done
getting AP scan result JSON ..done
{ "aps":[ {"i":0,"c":1,"m":"e0:28:6d:39:15:c0","ss":"TBU_ZERO","r":-91,"e":4,"se":0},{"i":1,"c":9,"m":"2c:30:33:ce:1d:ef","ss":"TBU","r":-85,"e":4,"se":0}] }
getting AP scan result JSON ..done
seect 0 - 1
getting AP scan result JSON ..done
getting attacks JSON...done 
starting deauth attack
getting attacks JSON...done 
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
running deauth attack done
running deauth attack done
getting attacks JSON...done 
running deauth attack done
stopping deauth attack
getting attacks JSON...done

H1ghTech, 2017-02-27:
somehow it did not disconnect this time but it still shows 0 pkts/s

spacehuhn, 2017-02-27:
@H1ghTech have you used the right SDK version? If the deauth attack isnt working it's probably beacause the necessary function can't be used. That's not an issue related to the testing branch, but when it crashes it could be something else - if that happens please paste the log here (but via pastebin or somehing else please).

@whyisthisusernamealreadytaken I will think about that one!

Beacuse the APScan bug:
I tested it and it looks like the json string can still get to long and the ESP crashes. The max max size seems to be around 3000 chars. I set maxAPScanResults to 30 (what equals a max json size of ~2870 chars) and it works.

sounakdeb, 2017-02-27:
Please add multiple wifi networks selection option.

H1ghTech, 2017-02-27:
@spacehuhn yes I have the right SDK and it does deauth the clients it's just that it doesn't show the packets

sounakdeb, 2017-02-27:
please add multiple wifi selection option selection option

sounakdeb, 2017-02-27:
please add multiple wifi selection option

H1ghTech, 2017-02-27:
@sounakdeb you can! Just select more than one Ap, or did you mean something else.

sounakdeb, 2017-02-27:
ya more than one Ap

spacehuhn, 2017-02-27:
@sounakdeb stop asking, multuple AP selection is implemented and I will delete all comments here which aren't about the testing branch. Please understand that.

sounakdeb, 2017-02-27:
ok

sounakdeb, 2017-02-27:
help me

nisc3, 2017-02-28:
New version works fine. After clicking 'start' on Attack panel, I get disconnected and reconnects to it (not a reboot)

DushyantPUNK, 2017-02-28:
This is still not working in my symbian  mobile,I have tried all the browsers.

Splodgie, 2017-02-28:
Hi,
 AP:     scan works 
Client:   Not working correctly.
Attack:   works (Will now Block all my device, The range issue of 55% is not a problem any more). all the new Attack options seem to work OK
Settings: work.

Client Problem.
If I use the deauhter without using the Client option, it all works OK and dose not fail.
If I use the Client option to scan the selected AP it fails 95% of the time. returning 0 Clients  (When I have  3 connected).
The Client scan can clear all the AP list and Fail, But staying in Clients and despite they being No AP list, the scanning option still works, but obviously always fails to find any Clients because the AP list is not populated. 
Going back to the AP option shows No AP's (the ESP has crashed). Although you can scan for AP's again, it will complete the scan and return found a list of AP found, The Client scan will never return any clients found, as the ESP needs to be reset/rebooted.
In the Client option with the scan time set to 15s, it may not scan for the full 15s it will stop scanning at 6s reload the scan page after reconnecting with O Clients, repeating the scan again, it never finds any clients, The AP list will still be showing a list of AP's but you have to rescan for AP's again, and the clients scan may work.
When the client scan crashes the ESP it some times puts the ESP into Flash mode and it locked up, so it has to be rebooted.
The only problem I have now is the Client Option.

DushyantPUNK, 2017-02-28:
THis is not working in symbian mobile browsers. 
I have a Nokia 5800 and in it none of the browser this is working.
In my pc this is working fine with no error except Clients Page.
Please make it available for symbian too.

spacehuhn, 2017-02-28:
The symbian plattform is outdated since 2012. It's probably not working because it doens't supports current JavaScript features or JS at all. Don't expect that this will ever work with it, sorry.

About your other issues -> I need the logs!
If the log seems ok use the developer options in your browser to see if the JS produces an error.

Thanks for your reports so far :)

nisc3, 2017-03-01:
Client scan not working.

Splodgie, 2017-03-01:
Client Scanning.
if the scanning is set to 15s, the scan will run for 7s then drop the WiFi connection, but show it's scanning for the remaining 8s, Then reconnect No Clients found.

if the scanning is set to 25s, the scan will run for 7s then drop the WiFi connection, but show it's scanning for the remaining 18s, Then reconnect, No Clients found.

if the scanning is set 2s, the scan will still run for 7s, then reconnect.   No Clients found

So the Client scan always runs for 7s, regardless of the time set and saved, after the 7s the WiFi connection is propped the remaining time if any is just waisted, until reconnect.

tpmodding, 2017-03-01:
works very good now! i can deauth my macbook, the new ssid list vor beacon is very nice! you can merge it to master :)

joneroy, 2017-03-02:
@spacehuhn  
Latest Testing Branch - 2/3/17 5.25 pm indian time
NodeMCU ESP8266 and WeMos D1 Mini
Android/Chrome

All devices working great deauthing my New HP 15-ab205TX ,Desktop, Old Laptop HP HP Pavilion G6-1003TX, Lenovo Mobile, Raspberry Pi 2 Model B successfully working fine.

1. Error: web page loading problem.
2. deauth working late press start button afer 5 sec later its working.
3. clients page sometime not opening.

Splodgie, 2017-03-03:
@spacehuhn 
Hi, Just installed the new commit Mar 3, 2017, Prevent user to clone the beacons of a hidden APs.
Installed as 'False' it works as before, Client scan Not working 90% of the time, seems like a small improvement.

Installed as 'true'  AP scan will not run, Flashes to scanning for 0.01s every time scan is pressed.
Networks Found 0, ( ESP is responding to the Scan request).

SSID: pwned
SSID length: 5
password: deauther
password length: 8
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 8
attack SSID encrypted: 0
nameList cleared

starting...
starting WiFi AP
SSID: pwned
Password: deauther
generating Macs...done
starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID - Hidden
scan done
getting AP scan result JSON done
{ "aps":[ ] }
getting AP scan result JSON done

Dave

spacehuhn, 2017-03-03:
The installation steps has changed. Follow the steps from the testing branch readme and it should work.

Splodgie, 2017-03-03:
Sorry, Looking into it now.

tpmodding, 2017-03-03:
@spacehuhn tested the new testing-branch, works...ap-scan works, client-scan works and i was able to deauth my macbook, beacon-clone and list works too :)

spacehuhn, 2017-03-03:
Just added a few more debug infos including to print the generated json and the client scan time. If no one has something against it I would like to procede to merge this into the master branch. Since the only bugs you reported so far were not reproucable for me and most likely because of browser (or human ;) ) incompatibilities.

Splodgie, 2017-03-03:
Hi, God you work fast.
Found my mistake. I have Arduino15 twice, In Local and Roaming. I had to put the ESP8266WiFi.cpp and ESP8266WiFi.h in both, then it worked OK.
Just loaded the latest commit, and a quick test shows it's working OK'ish, Still cannot get Client scan to work, maybe once in every 15 full resets, The attack still works, both on True & false settings.
Tested on all my ESP,s and it still loads on ESP-01 1M  SPIFFS 256K.
Dave

Splodgie, 2017-03-03:
Hi, 73390c6  Continuous Works OK.

poslogic, 2017-03-03:
Just tested latest testing branch :+1: 
NodeMCU dev
Android/Chrome

All good and working great, deauthing everything on my network very effectively.
All new features working and seeing hidden networks, thank you @N0vaPixel
 
The continuous scan button seems to stop after 1 minute, not sure if this is designed to happen but not an issue for me.

Great work @spacehuhn seems good to merge for me :+1: 
Will continue testing with other boards.

spacehuhn, 2017-03-03:
yeah the continuous scan will eventually stop at some point when a timeout occurs. That's just something which can't be changed the way it's implemented but still better then nothing.
Good to hear that everything works for you! :)

Robiii97, 2017-03-03:
When i change the SSID and Password in settings reboot it my connection fails and im unable to connect to the ESP.
I also tried changing the SSID and password in adruino before uploading but that gives me the same crashing result.

joneroy, 2017-03-03:
@spacehuhn 

Latest Testing Branch - 4/3/17 1.24 am (india time)
NodeMCU ESP8266 and WeMos D1 Mini
Android/Chrome

Every feature test all my devices working great : New HP 15-ab205TX-/-Desktop-/-Old Laptop HP HP Pavilion G6-1003TX-/-Lenovo a916 Mobile-/-Lumia 730 Mobile-/-Sony xperia z2 Mobile-/-Intex Mobile-/-Raspberry Pi 2 Model B-/-successfully working fine.

# **Bugs:**

1.# **Error: web page loading problem. // "clients.html" only.**
[image]

2.# **setting box values blank**
[image]

tobozo, 2017-03-03:
Just cloned the testing version and got this error while trying to conpile : 

```
sketch/APScan.cpp: In member function 'bool APScan::start()':
APScan.cpp:14: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'
     results = WiFi.scanNetworks(false, true); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)
                                            ^
sketch/APScan.cpp:14:44: note: candidate is:
In file included from sketch/APScan.h:6:0,
                 from sketch/APScan.cpp:1:
/home/tobozo/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)
     int8_t scanNetworks(bool async = false);
            ^
/home/tobozo/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided
exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'

```

Changin `results = WiFi.scanNetworks(false, true);` by `results = WiFi.scanNetworks();` seems to fix the problem but I've got a feeling this is going backwards in terms of features.

Do I need to change something to my Wifi library in order to get this working ? I thought Async Wifi scan only happened in an earlier version of the SDK and was not available yet in the 2.0.0

spacehuhn, 2017-03-03:
OK then here again.. :D 
read the readme - it has changed!

joneroy, 2017-03-03:
@tobozo 

Copy these files and past them at the location.

# **Use sdk_fix files it's updated testing branch**

user_interface 				--->  \packages\esp8266\hardware\esp8266\2.0.0\tools\sdk\include
ESP8266WiFi.cpp and ESP8266WiFi.h 	--->  \packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src

tobozo, 2017-03-03:
thx, nice fix working just fine now

N0vaPixel, 2017-03-03:
Update 03/03/2017

Scan for hidden APs (with settings)
Hide and set the channel of your own AP (with settings)

Please test and send me your feedbacks #70

Splodgie, 2017-03-04:
The New Hide DeAuther IP works OK, (I use a WiFi Manager app on my Android phone, It made this buggy, I now have to forget and Add WiFi networks, both in the App and on the phone's normal WiFi list, or it will not stay connected to the ESP, never had to do this before.) Just the way I use my Phone.   Thanks for all your Brilliant work spacehuhn.
Dave

spacehuhn, 2017-03-04:
It's merged! Thank you all for your support! 👍 

If there's still a problem you can open up a new issue.
I'll let the testing branch coexist for future updates and testing.

closed 2017-03-04 by spacehuhn