SpacehuhnTech/esp8266_deauther #44: Hast du schon einmal über eine Projektseite nachgedacht + Blog

H1ghTech, 2017-02-27:
Hey :) 
Ich finde deine Projekte echt super und wollte dir deswegen gerne eine Projektseite machen z.B auf meinem Server oder auch auf github-pages. 

Ach so: ich habe Repos, die sind nur alle private lol ( nur um das klarzustellen weil ich das oft gefragt werde

spacehuhn, 2017-02-27:
Coole Idee.
Wenn du darauf lust hast gerne!
Ich würde dann allerdings bevorzugen das ganze auf github pages zu machen und auch auf englisch und so das jeder mithelfen kann. Also so als richtiges open source Projekt.

Wenn du twitter hast kannst du mich da auch gerne bei mir melden (wie hier @spacehuhn).

H1ghTech, 2017-02-28:
OK welche Domain möchtest du?
[image]

spacehuhn, 2017-02-28:
was genau hast du denn jetzt vor? Einen Blog habe ich ja schon.
Wenn du eine allgemeine Projektseite machen möchtest gerne, aber dann musst du mir auch sagen was du dir genau darunter vorstellst.

H1ghTech, 2017-02-28:
Ja ne Projektseite wo halt all deine Projekte erklärt werden vlt auch mit wiki

spacehuhn, 2017-02-28:
Dafür hab ich ja schon Spacehuhn.de
Ich wäre wenn für eine Projektseite über den deauther. Wiki wäre da wahrscheinlich etwas übertrieben aber mit genaueren Anleitungen, Informationen, Bildern etc. wäre das wahrscheinlich schon ganz nützlich.

H1ghTech, 2017-02-28:
OK kann Ich machen

H1ghTech, 2017-02-28:
Ich lass die Seite dann erstmal bei mir laufen und du kannst dir die dann einfach forken

tpmodding, 2017-02-28:
das wird nix...irgendwann hast du keine lust mehr zu supporten...lieber das github wiki nutzen

spacehuhn, 2017-02-28:
Ich überlass das euch. Ich habe nur keine Zeit mich auch noch darum zu kümmern.

H1ghTech, 2017-02-28:
@tpmodding Ich mach das mit nem statischen site gen
der läuft in gh-pages (jekyll)
wenn ich net mehr supporte kann @spacehuhn das auch ganz einfach selber machen

H1ghTech, 2017-02-28:
https://github.com/H1ghTech/esp kannste ja mal anschauen =)

spacehuhn, 2017-03-02:
Für die Übersicht schließe ich das jetzt mal, ist ja kein Issue.

closed 2017-03-02 by spacehuhn

H1ghTech, 2017-03-02:
@spacehuhn OK ich hab die Seite ja fertig wie ist das denn jetzt willste die haben?

spacehuhn, 2017-03-02:
Hatte noch keine Zeit sie mir anzuschauen.

H1ghTech, 2017-03-02:
Ah ok du kannst ja Mal ne neue repo machen mit dem Namen spacehuhn.github.io

H1ghTech, 2017-03-03:
Uhm?!?

spacehuhn, 2017-03-04:
Also habs mir angeschaut. Du hast ja jetzt eine Projektübersicht von meinem Account gemacht. Das ist eigentlich etwas was ich auf meiner Website noch selber machen werde und ich sehe jetzt keinen Grund 2 Seiten dafür online zu stellen.

Hatte das jetzt eher auf dieses Projekt bezogen verstanden. Was aber auch unsinnig ist wenn ich so drüber nachdenke. Wie @tpmodding sagte wäre das github Wiki da sinnvoller. Github selber ist ja auch schon eine Projektseite also nicht böse gemeint aber ich brauche es ganz einfach nicht.

H1ghTech, 2017-03-04:
OK gut