SpacehuhnTech/esp8266_deauther #46: [Feature request] Show hidden ssid's

H1ghTech, 2017-02-28:
Almost every network in my house are not broadcasting the SSID (for obvious reasons). So I asked myself if it is possible to show them :)

spacehuhn, 2017-02-28:
I think it's possible but I haven't tried it out yet. It would require to get into the promiscuous mode just like when it's scanning for clients and it would only see the MAC adress.

joneroy, 2017-02-28:
@H1ghTech ya it's possible to show hidden ssid's using esp8266 but i don't no software implementation easy or not.

DevShekhawat7, 2017-03-03:
ya please add this feature..there are so many hidden ssid around me which this guy is no pulling up. please add this feature

spacehuhn, 2017-03-03:
merged into testing branch. You can test it now.

DevShekhawat7, 2017-03-03:
@spacehuhn thank you soo much

H1ghTech, 2017-03-03:
@spacehuhn thank you:)

closed 2017-03-03 by spacehuhn