SpacehuhnTech/esp8266_deauther #48: Sending 0 packets/second

wnew, 2017-03-01:
When I deauth all or deauth selected APs. The status reads 0ptks/s and my wifi devices maintain their connections.

Could this be a bug or an I doing something wrong.

poslogic, 2017-03-01:
Deauth attack won't work

If you see 0 pkts/s on the website you have made a mistake. Check if you have followed the the installation steps correctly and that the right SDK installed, it must be version 2.0.0!
If it can send packets but your target don't loose its connection then the WiFi router uses 802.11w and it's protected against such attacks or they communicate via 5 GHz WiFi, which the ESP8266 doesn't support.

H1ghTech, 2017-03-01:
In the testing branch go to settings and set the pkts/s after that it worked for me

wnew, 2017-03-02:
Thanks guys,

It turns out I have version 1.5.3 of the SDK. Now to try and upgrade it.

closed 2017-03-02 by wnew