SpacehuhnTech/esp8266_deauther #49: Deauth doesn't work

JonathanJhon55, 2017-03-01:
So I purchased this board, only to use this script.

Atlough the death shows 20pkt/s nothing happens.

I am not on 5Ghz. And it did work on my Raspberry Pi with the same network.

Any advice? I need to get this working. Thanks

poslogic, 2017-03-01:
Try using the latest testing branch this is working very well for me.
https://github.com/spacehuhn/esp8266_deauther/tree/testing

JonathanJhon55, 2017-03-01:
@poslogic I just tested it. Also with 200 packets, it doesn't do anything. My WiFi doesn't even lagg..

poslogic, 2017-03-01:
Do you have access to any other AP's to test on? Friends or mobile phone hotspot?

I have successfully tested this on my own WiFi, my mums and my Android phone hotspot.

What board are you using?

JonathanJhon55, 2017-03-01:
https://www.amazon.co.uk/NodeMCU-Internet-Development-ESP8266-ESP-12E/dp/B01GCK3J40/ref=sr_1_1?ie=UTF8&qid=1488393656&sr=8-1&keywords=tinxi+esp8266

I'm using this board. Same seller, just from amazon.fr.

I just tested it too. It doesn't do anything also

poslogic, 2017-03-01:
I have the same board.
Struggling to think of reasons why yours does not work.

Double check that you followed all steps in Readme.
Make sure board is correctly selected in Arduino/Tools

Board - NodeMCU 1.0 (Esp 12-E)
CPU frequency 80 MHZ
Flash Size 4m
Upload speed 115200
Programmer ArduinoISP

joneroy, 2017-03-01:
You are using the right SDK version.

poslogic, 2017-03-01:
Are you seeing any errors in the Serial Monitor?

Are you seeing (running deauth attack done) when deauthing?

JonathanJhon55, 2017-03-01:
@poslogic

poslogic, 2017-03-01:
@JonathanJhon55 Serial Monitor output looks the same as mine, only other thing I can think of is that your WiFi router uses 802.11w??? Maybe switch encryption to none and try?

JonathanJhon55, 2017-03-01:
@poslogic See this screenshots. It doesn't use 802.11w 😞 

[image]

joneroy, 2017-03-01:
@poslogic  My Serial Monitor not working only show 3 line :

starting...
SSID: pwned
Password: deauther

joneroy, 2017-03-01:
@JonathanJhon55  You using wifi ddos protection feature on your wifi.

poslogic, 2017-03-01:
@joneroy What board are you using? 

Have you switched to the correct SDK 2.0.0

Have you added the lines of code to the user_interfaces.h file (and clicked on save aftwards)

Mcewen123, 2017-03-01:
Hi
Try using Board - NodeMCU 0.9 (Esp 12-E) it works for me, I have the NodeMCU 1.0 (Esp 12-E) but.................everything works but the deauthing with Board -NodeMCU 1.0 (Esp 12-E) , when I choose the Board -NodeMCU 0.9 (Esp 12-E) everything works, give it a try

joneroy, 2017-03-01:
@poslogic  i'm using correct SDK 2.0.0 and added the lines of code to the user_interfaces.h file but My Serial Monitor not working only show 3 line :

starting...
SSID: pwned
Password: deauther

i'm using this 115200 baud

[image]
[image]

esp8266_deauther working but i dont show Serial Monitor.

poslogic, 2017-03-01:
@joneroy We are all using the testing branch with debugging on.

Are you using the Master or Testing branch?

joneroy, 2017-03-01:
@poslogic  Master branch

poslogic, 2017-03-01:
@joneroy Try the testing branch if you like it has new features and I think it will be merged into the master soon.

Remember to keep a copy of the version that you have working now, just in case ;-)

https://github.com/spacehuhn/esp8266_deauther/tree/testing

joneroy, 2017-03-01:
@poslogic  I'm trying the testing branch and leave feedback for Stefan.

DevShekhawat7, 2017-03-03:
Correct me if i am wrong, If someone is connected with LAN cable and if we deauth the network, will it work? because in my case it's not happening

poslogic, 2017-03-03:
No this is for WiFi 802:11b,g,n - 2.4 GHz only

Check the other projects by Stefan for LAN

N0vaPixel, 2017-03-03:
The WiFi deauth attack doesn't affect LAN clients.
This attack spoof clients by sending IEEE 802.11 packets

spacehuhn, 2017-03-04:
Can we close this?

joneroy, 2017-03-04:
@spacehuhn close this issue it's solved

closed 2017-03-04 by spacehuhn

emon11, 2019-04-30:
> @spacehuhn close this issue it's solved

how please

spacehuhn, 2022-02-06:
Hi, I know this issue is old and has been closed, but since it ranks well on Google, I would like to point people to a new blog post that discusses this topic in further detail:
https://blog.spacehuhn.com/deauth-attack-not-working/