SpacehuhnTech/esp8266_deauther #50: error: espcomm_open failed

JonasZaverka, 2017-03-02:
I thing I have my code to work but it won't upload 

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

I am a starting with programing and stuff like that.
thx for help Jonas

poslogic, 2017-03-02:
Have you followed all steps in the Readme exactly?
https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md

Please provide more information E,g: Board type, SDK version, Arduino/Tools settings, debugging steps that you have already tried.

JonasZaverka, 2017-03-02:
Yes I do everything like in readme I try different board types
but I think problem is in my   A+ nodeMcu Lua ESP8266 ESP_12E ch340g WIFI
Internet development board module
I try different payloads but I have the same issue I both this board main
for this project (wifi jammer).
Thx for your answer Jonas

JonasZaverka, 2017-03-02:
[image]

I try difrent boards, programers. 
Sorry fro screenshot in different language

JonasZaverka, 2017-03-02:
LED on my board make me think the problem is in board.
Somthing like i have to flash it.

poslogic, 2017-03-02:
Try changing board type to Board - NodeMCU 1.0 (Esp 12-E)

Then try using the testing branch of this.... https://github.com/spacehuhn/esp8266_deauther/tree/testing

If it compiles and runs then
Open the Tools/ Serial Monitor in Arduino and report any errors that you see.

poslogic, 2017-03-02:
Flash your board with a simple example sketch to test communication with board first before trying this level.

JonasZaverka, 2017-03-02:
I try change board type and upload the blink example but

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

Thx for your work

poslogic, 2017-03-02:
It is not my work it is @spacehuhn who is developer.

Did blink example work on your board?

JonasZaverka, 2017-03-02:
No i get
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

poslogic, 2017-03-02:
Okay you need to be able to communicate with your board before trying to flash this programme.

hope this video helps, I need to sleep now but will be back tomorrow, Good Luck :-)
https://www.youtube.com/watch?v=G6CqvhXpBKM

joneroy, 2017-03-02:
@JonasZaverka  if you are using nodeMcu Lua ESP8266 ESP_12E board using this driver "CH341SER" solve your error: espcomm_open failed and select second port and then press "get board info" i hope it's working.

Splodgie, 2017-03-02:
It looks like your board is not going into flash mode.   Check that the 2 buttons are not stuck and are clicking when you press them, you can put this board into flash mode by connecting  GPIO_02 to VCC,   and GPIO_15 to GND. 
I found this link that some one created for this issue 
https://github.com/esp8266/Arduino/blob/master/doc/faq/a01-espcomm_sync-failed.md
(A Very good piece of work).

tpmodding, 2017-03-02:
https://youtu.be/dgw8tnPFGSY

H1ghTech, 2017-03-02:
Do you have a Mac?

tpmodding, 2017-03-02:
this is a general esp problem, you could close this issue and continue discussing

closed 2017-03-02 by spacehuhn

sahanwa, 2017-12-06:
Please refer this link.. You might find your answer here sometimes.. ESP Worked for me after trying for around a week with it... 
https://github.com/esp8266/Arduino/issues/466