SpacehuhnTech/esp8266_deauther #51: Error: web page loading problem

joneroy, 2017-03-02:
Error: web page loading problem only on 2 page 

1.clients page // when press start button "Error: web page loading problem"
2.attack page // when press start button deauth "Error: web page loading problem"

nisc3, 2017-03-02:
Yes, you get disconnected and reconnect to it and both deauthentication and client scan works fine. It is because of low memory (?)

joneroy, 2017-03-02:
@nisc3  I think It's not low memory prob.

nisc3, 2017-03-02:
Yep, I agree. And as you said client scan doesn't work properly.

joneroy, 2017-03-02:
two prob i found :

1.clients page // when press start button "Error: web page loading problem"
2.attack page // when press start button deauth "Error: web page loading problem"

joneroy, 2017-03-02:
@nisc3  i think it's code prob master branch it's workin fine

joneroy, 2017-03-02:
[image]

spacehuhn, 2017-03-03:
like I said many times before: look at the logs and use the dev tools of your browser. Otherwise we can't help you. 

It's like you ask why your car isn't starting and sending a picture of it. I have no idea...

joneroy, 2017-03-04:
@spacehuhn Web Page Loading Problem is solved testing branch

closed 2017-03-04 by spacehuhn