SpacehuhnTech/esp8266_deauther #54: Add a feature to preselect wifis

Hammad1029, 2017-03-02:
Can you add a feature to preselect WiFi ssids? like if I know the ssid of the WiFi of the place I'm going to, I'll select it before leaving and it'll start spamming deauth packets to that WiFi. Whether by waiting to detect the specific ssid or spamming deauth packets dedicated to that WiFi before getting in its vicinity. Is it even possible? if yes can you add it please

joneroy, 2017-03-02:
it's possible

Skulkx, 2017-03-16:
I have a Feature Request which is actually quite similar to this one.
I would love to have the option to automatically start an Attack on PowerOn.

So that i can set up, lets say, a Beacon-Jammer-Attack, at home. Then, at the site, i'll power the ESP8266 via Battery and the defined Attack would launch

Thank you very much for your precious time :-)

closed 2017-04-02 by spacehuhn