SpacehuhnTech/esp8266_deauther #56: [Feature request] specifies connected ap __ clients

joneroy, 2017-03-02:
@spacehuhn 
[image]

joneroy, 2017-03-02:
[image]

joneroy, 2017-03-02:
please solve this prob

poslogic, 2017-03-02:
You asked before if you could use and change code to your needs.

You can, please do and share your results.

joneroy, 2017-03-02:
@poslogic can you please add this option

spacehuhn, 2017-03-03:
1. I don't get what you want
2. like poslogic said: implement it yourself
or pay me per hour...

joneroy, 2017-03-03:
after select 2 ap the client result is mix. please specify ap clients mean A&B is ap and i select both but client result time show A&B ap mix client result like.

**# AP**

A is wifi>1client.5client.3client

B is wifi>6client.7client.10client.11client

**# client** result now show :

1.client- jone
5.client- alex
3.client- shreen
6.client- kid
7.client- sean
10.client- chris
11.client- stife

after you add new this feature

**# AP**

A is wifi>1client.5client.3client

B is wifi>6client.7client.10client.11client

**# client** result after :

A:
1.client- jone
5.client- alex
3.client- shreen
B:
6.client- kid
7.client- sean
10.client- chris
11.client- stife

@spacehuhn feature like this can you add this feature.

tpmodding, 2017-03-03:
der will das du bei den clients anzeigst welcher client zu welchem ap gehört...welcome to github mit script kiddies... die wollen jetzt alles sofort haben :) du stumpfst schon ab mit der zeit, kein stress :D

N0vaPixel, 2017-03-08:
Added in the testing branch

closed 2017-03-08 by N0vaPixel