SpacehuhnTech/esp8266_deauther #57: [Feature request] Login Page

joneroy, 2017-03-02:
@spacehuhn 
add a login page 192.168.4.1 
>user name
>password

login

then redirected to main ap page.

H1ghTech, 2017-03-02:
Wow you have really good ideas!!!

joneroy, 2017-03-02:
@H1ghTech  I like open source projects. thank you for like my ideas!!

H1ghTech, 2017-03-02:
Np xD

poslogic, 2017-03-02:
Logging into the board via WiFi connection requires a password.

Default:
pwned
Deauther

Which can be changed.
What would login page add?

joneroy, 2017-03-02:
wifi password hacking now day super easy. 2nd time authentication needs like a router.

poslogic, 2017-03-02:
Okay another level of authentication is always good.

But using good passphrase and limited time that (I personally use) ESP board is on for makes attacks not a problem ( in my use for this project)

spacehuhn, 2017-03-02:
I agree with poslogic. This is obsolete, it has already an authenticstion build in which is (mostly) annoying enough.

closed 2017-03-02 by spacehuhn