SpacehuhnTech/esp8266_deauther #58: compiling issue

TwinkleFredrik, 2017-03-02:
Arduino: 1.6.13 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

Build options changed, rebuilding all
sketch\Attack.cpp: In member function 'bool Attack::send(uint8_t*, int)':

Attack.cpp:41: error: 'wifi_send_pkt_freedom' was not declared in this scope

   if(wifi_send_pkt_freedom(buf, len, 0) == -1){

                                       ^

sketch\Attack.cpp: In member function 'void Attack::run()':

Attack.cpp:179: error: 'wifi_send_pkt_freedom' was not declared in this scope

       if(wifi_send_pkt_freedom(deauthPacket, 26, 0) == -1){/*

                                                   ^

Attack.cpp:190: error: 'wifi_send_pkt_freedom' was not declared in this scope

       if(wifi_send_pkt_freedom(deauthPacket, 26, 0) == -1){/*

                                                   ^

Attack.cpp:221: error: 'wifi_send_pkt_freedom' was not declared in this scope

         if(wifi_send_pkt_freedom(packet, packetSize, 0) == -1){/*

                                                       ^

exit status 1
'wifi_send_pkt_freedom' was not declared in this scope

TwinkleFredrik, 2017-03-02:
Update: new problem..
---
Arduino: 1.6.13 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

c:/users/twinkle-fredrik/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\TWINKL~1\AppData\Local\Temp\arduino_build_51771/esp8266_deauther.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.

poslogic, 2017-03-02:
Looks like you are either not using the SDK 2.0.0 version or you haven't added the few lines of code to the user_interfaces.h file? And or clicked save aftwards.

Please double check through the reader.

poslogic, 2017-03-02:
Also updating Arduino software may help? 

The latest version is 1.8.1

https://www.arduino.cc/en/Main/Software

joneroy, 2017-03-02:
@TwinkleFredrik read this https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md

TwinkleFredrik, 2017-03-02:
I updated to 1.8.1
[image]
I clicked save
[image]
I have the correct version

I am still receiving the same error message

TwinkleFredrik, 2017-03-02:
@joneroy That is the page I am reading from so why tell me to read it when I am. I keep checking if I did something wrong but cant find the issue

joneroy, 2017-03-02:
@TwinkleFredrik if you are using nodeMcu Lua ESP8266 ESP_12E board using this driver "CH341SER" #50

poslogic, 2017-03-02:
Make sure version 2.0.0 is selected then close, reboot, flash simple sketch and try again.

joneroy, 2017-03-02:
@poslogic he use sdk 2.0.0

TwinkleFredrik, 2017-03-02:
It was the board selection, what @joneroy hinted to.. Thanks joneroy!
[image]
Thanks for the help. If I need to close this thread... how xD

joneroy, 2017-03-02:
@TwinkleFredrik Happy Hacking!!

TwinkleFredrik, 2017-03-02:
@joneroy well.. its for educational properties to just learn how people use this maliciously since I have a neighbor that keeps doing it..

poslogic, 2017-03-02:
@TwinkleFredrik glad you got it working, check out the testing branch for the latest features ;-)

@joneroy 👍

closed 2017-03-02 by TwinkleFredrik