SpacehuhnTech/esp8266_deauther #59: [Feature request] Add an option to automatically scan

A4A4ANDY, 2017-03-02:
Not that big of a life changer I think but would be nice :)

N0vaPixel, 2017-03-03:
Scan for AP or for client ?
And why since you need to select yourself the targets.

nisc3, 2017-03-03:
Automatic scan for AP is a nice idea!

spacehuhn, 2017-03-03:
I don't know If this will work out... Possible but a lot of work! With the current structure it isn't possible, you'd have to change so much in order to get this to work!
So it's way easier just to click on the little scan button.

joneroy, 2017-03-03:
@spacehuhn I think little **scan** button better then auto ap scan.

spacehuhn, 2017-03-03:
But I think this can be achieved with JavScript directly in the webinterface. Then the main code remains untouched. I'll test this.

spacehuhn, 2017-03-03:
There you have it: d63c97f06a8b3527fd476b800bf43bb7f8bb636b

closed 2017-03-03 by spacehuhn

joneroy, 2017-03-03:
@spacehuhn Grate!!