SpacehuhnTech/esp8266_deauther #62: APs scan issue

TwinkleFredrik, 2017-03-03:
I had a working board that worked flawlessly with the "master" version.. but after giving the "esp8266_deauther-testing" version a try the only message I get now when trying to scan for APs is
[image]

I do not understand what has happened?

poslogic, 2017-03-03:
Disable/forget your home network so that the phone/PC (device you are using) can only automatically connect to the board.

TwinkleFredrik, 2017-03-03:
The device I am using to connect to it has none remembered so that is not the problem.. It is connected but after failing a couple of times my device becomes disconnected.

poslogic, 2017-03-03:
Flash a simple blink sketch to your board, then try removing and rebooting your PC.

Then try re- flashing the master branch that works well for you.

Hope this helps.

TwinkleFredrik, 2017-03-03:
I uploaded 
[image]
This script works BUT... The other one seems to be running as well in the background? The old ssid I chose and password pops up in my wifi list.

TwinkleFredrik, 2017-03-03:
I tried to upload the master version again but problem still coexists :(

tpmodding, 2017-03-03:
no infos...which phone? which browser? 

I connect to my NodeMCU with my iPhone 6, if I scan AP's it does not disconnect, but if i scan for clients, my iphone disconnects, cause the esp scans, in this time my phone connects to my local router -> so i have to reconnect to my nodemcu, than it shows me the clients

TwinkleFredrik, 2017-03-03:
@tpmodding you must have misunderstood... I am not using any phone and it would be irrelevant when I have already tested on sevral browsers and devices. Nothing says the iphone is better at connecting to the internet compared to any android that may be more intelligent. and if you are wondering.. No, I am not using internet explorer....

poslogic, 2017-03-03:
Try clearing browse cache and cookies.

TwinkleFredrik, 2017-03-03:
no progress.. same issue

tpmodding, 2017-03-03:
just wanted to help and you give me a thumbs down? you say you are not using IE but dont say which one you use... no info, no support

TwinkleFredrik, 2017-03-03:
@tpmodding Just told you I try what I can.. Chrome, Opera, firefox, safari, chromium and edge. happy?

TwinkleFredrik, 2017-03-03:
Update: If it has something to do with it, the webserver crashes and turns everything white when I click the clients tab

joneroy, 2017-03-03:
@TwinkleFredrik i think it's a bug on ap.html #51 press hastag check the prob and say it's same prob or not

TwinkleFredrik, 2017-03-04:
@joneroy No my problem does not seem to be related to #51 The only error message I get is the one you see on the first post. The page loads just fine and everything seems to work but when I press scan on the APs tab the esp8266 crashes and gives me that until I unplug it from power.

TwinkleFredrik, 2017-03-04:
@spacehuhn what is this duplicate to?

spacehuhn, 2017-03-04:
#5

joneroy, 2017-03-04:
@TwinkleFredrik you are using testing branch or not so you are not using testing branch use it no bug found everything solved by @spacehuhn

joneroy, 2017-03-04:
@spacehuhn 

# **close this issue it's solved**

TwinkleFredrik, 2017-03-04:
But its not solved when my board isnt working propperly and I dont know what to do other than searching around to finally find the solution somewhere on this page :/

TwinkleFredrik, 2017-03-04:
Or has it recently been updated and I need to download both of the versions and try again?

joneroy, 2017-03-04:
@TwinkleFredrik Download both of the versions and try again.

tobozo, 2017-03-04:
@TwinkleFredrik also check the recent changes in the ReadMe

TwinkleFredrik, 2017-03-04:
I tried the updated master build.. got this: Arduino: 1.8.1 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

sketch\APScan.cpp: In member function 'bool APScan::start()':

APScan.cpp:14: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

     results = WiFi.scanNetworks(false, settings.apScanHidden); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)

                                                             ^

sketch\APScan.cpp:14:61: note: candidate is:

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\Twinkle-Fredrik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\Users\Twinkle-Fredrik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

TwinkleFredrik, 2017-03-05:
Should I just open a new issue post with the comment above? This is during compilation..

spacehuhn, 2017-03-05:
> @TwinkleFredrik also check the recent changes in the ReadMe

People trying to help you, just listen to them and your problem is solved.

TwinkleFredrik, 2017-03-05:
But how do I find the recent changes? I search for recent changes but cant find out what is different with the readme file.

spacehuhn, 2017-03-05:
dude... read the readme!

joneroy, 2017-03-05:
@TwinkleFredrik 

https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md

TwinkleFredrik, 2017-03-05:
Thank you @joneroy finally works again! :)
You really are good at helping.

closed 2017-03-05 by TwinkleFredrik