SpacehuhnTech/esp8266_deauther #64: error: 'wifi_send_pkt_freedom' was not declared

db306, 2017-03-03:
Hi there,

I'm having this issue while compiling. Any luck someone knows what I'm doing wrong or what is going on here ? Seems like there are some dependencies issues to me

thanks

Arduino: 1.6.7 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

Build options changed, rebuilding all
sketch/Attack.cpp: In member function 'bool Attack::send(uint8_t*, int)':
Attack.cpp:41: error: 'wifi_send_pkt_freedom' was not declared in this scope
   if(wifi_send_pkt_freedom(buf, len, 0) == -1){
                                       ^
sketch/Attack.cpp: In member function 'void Attack::run()':
Attack.cpp:179: error: 'wifi_send_pkt_freedom' was not declared in this scope
       if(wifi_send_pkt_freedom(deauthPacket, 26, 0) == -1){/*
                                                   ^
Attack.cpp:190: error: 'wifi_send_pkt_freedom' was not declared in this scope
       if(wifi_send_pkt_freedom(deauthPacket, 26, 0) == -1){/*
                                                   ^
Attack.cpp:221: error: 'wifi_send_pkt_freedom' was not declared in this scope
         if(wifi_send_pkt_freedom(packet, packetSize, 0) == -1){/*
                                                       ^
exit status 1
'wifi_send_pkt_freedom' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

poslogic, 2017-03-03:
Do you know what board you are using?

You need SDK2.0.0
You need to add lines of code to user_interface.h file as per readme.

And if you're board has less than 1MB storage you need to edit the data.h file

joneroy, 2017-03-03:
@db306 

update ARDUINO 1.8.1 IDE https://www.arduino.cc/en/Main/Software

# **config:**

1 Install Arduino and open it.

2 Go to File > Preferences

3 Add http://arduino.esp8266.com/stable/package_esp8266com_index.json to the Additional Boards Manager URLs. (source: https://github.com/esp8266/Arduino)

4 Go to Tools > Board > Boards Manager

5 Type in esp8266

6 Select version 2.0.0 and click on Install (must be version 2.0.0!)

[image]

7 Go to File > Preferences

8 Open the folder path under More preferences can be edited directly in the file

[image]

9 Go to packages > esp8266 > hardware > esp8266 > 2.0.0 > tools > sdk > include

10 Open user_interface.h with a text editor

11 Scroll down and before #endif add following lines:

typedef void (*freedom_outside_cb_t)(uint8 status);
int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb);
void wifi_unregister_send_pkt_freedom_cb(void);
int wifi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);

[image]

don't forget to save!

12 Download and open esp8266_deauther > esp8266_deauther.ino in Arduino

13 Select your ESP8266 board at Tools > Board and the right port at Tools > Port
If no port shows up you may have to reinstall the drivers.

14 Upload!

Your ESP8266 Deauther is now ready!

# **Same prob solved press hastag** #58 

# **Happy Hacking!!**

joneroy, 2017-03-03:
@db306 

Hi there,

I'm having this issue while compiling. Any luck someone knows what I'm doing wrong or what is going on here ? Seems like there are some dependencies issues to me

thanks

Arduino: 1.6.7 (Mac OS X), Board: "# **Generic ESP8266 Module**, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

Build options changed, rebuilding all
sketch/Attack.cpp: In member function 'bool Attack::send(uint8_t*, int)':
Attack.cpp:41: error: 'wifi_send_pkt_freedom' was not declared in this scope
if(wifi_send_pkt_freedom(buf, len, 0) == -1){
^
sketch/Attack.cpp: In member function 'void Attack::run()':
Attack.cpp:179: error: 'wifi_send_pkt_freedom' was not declared in this scope
if(wifi_send_pkt_freedom(deauthPacket, 26, 0) == -1){/*
^
Attack.cpp:190: error: 'wifi_send_pkt_freedom' was not declared in this scope
if(wifi_send_pkt_freedom(deauthPacket, 26, 0) == -1){/*
^
Attack.cpp:221: error: 'wifi_send_pkt_freedom' was not declared in this scope
if(wifi_send_pkt_freedom(packet, packetSize, 0) == -1){/*
^
exit status 1
'wifi_send_pkt_freedom' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
enabled in File > Preferences.

# **Select Right Esp8266 board**

#  **if you are using "nodeMcu Lua ESP8266 ESP_12E" board using this driver "CH341SER"**

[image]

spacehuhn, 2017-03-03:
This is an obvious issue - you have to use the right SDK version and edit it like it's decribed in the readme.

closed 2017-03-03 by spacehuhn