SpacehuhnTech/esp8266_deauther #65: [Feature request] Add manifest.json

tobozo, 2017-03-03:
How about adding a server.on('/manifest.json', sentJSONManifest) with a full list of HTML/CSS/Js/image files available in data.h?

I imagine at least three benefits for that: 

  1) it would unlock the possibility to use service workers when in offline mode (browser and webview)
  2) it would enable persistence if using a webview in an Android app
  3) it could ease prefetching when using the incoming SerialServer feature (serial proxy, http bridge)

spacehuhn, 2017-03-03:
I personally hate cache manifests. Yes they sound promising but browsers normally caches files automatically and to force them to cache files makes development extremly annoying! Everytime you change something in these files you have to delete your browser cache to get it to load the new and changed file.
Just imagine how many people would struggle with that.

And for your android app, don't you use the html files anyway without using the webserver?

tobozo, 2017-03-03:
I hate cache manifests too, I'm more interested in the file list it can provide to the incoming SerialServer than the HTML5 features that come with it.

Being able to list the server files can ease a lot the pain for phonegap+cordovarduino, node+serialport, ruby+serialport-server as it removes the need for the bridge to act as a dummy proxy.

Also in another scenario (probably cache-nightmarish and without serial availability), the android app could eventually access 192.168.1.4 and make good use of service workers when the clientscan disconnects the browser from the webserver.

spacehuhn, 2017-03-03:
Ah ok so just as file list and cach is more optional.   
So it's just a json file with an array of all files?

tobozo, 2017-03-03:
yep, it doesn't even have to be named manifest.json

spacehuhn, 2017-03-03:
Ok with all json files like attackStart.json and other commands only used by the webserver?

tobozo, 2017-03-03:
Only the static files are necessary for the SerialServer needs, but if there's no specific cost to just send the full handlers list I'm sure a good use can be made of knowing the json file names toos.

Also a more philosophical question comes out of it: should this file list itself?

spacehuhn, 2017-03-03:
Let's do a list of listes and then a list for these listes, but who listes the list listed list?
... :D  jap that's true. 

OK so I just make an array for static file and one for the commands in the manifest. You can edit this if you need to, the main thing is that we then have a list with everything. 
damn...should I now include the manifest to itself?  You really got me thinking on this...

spacehuhn, 2017-03-04:
00ec81abd9aafdbbb872b88a02e225c3cf69931b ?

tobozo, 2017-03-04:
\o/

spacehuhn, 2017-03-04:
hurray 😄

closed 2017-03-04 by spacehuhn