SpacehuhnTech/esp8266_deauther #66: Add Password grabber for beacon span

DushyantPUNK, 2017-03-03:
Add Password grabber for beacon span so as to get the password of the ap that we have deauthed. This will be helpful for hacking wifi without any attacks.

spacehuhn, 2017-03-03:
"also implement to catch unicorns and scan for rare pokemons your the area."
Ok sarcasm aside...
What do you mean by password grabber? Do you know how Wi-Fi works?

DushyantPUNK, 2017-03-03:
you may conduct MITM attack and intercept the traffic. here you clone a network with your own and trick users into connecting into that network instead of the real one.

poslogic, 2017-03-03:
@spacehuhn Lol, just think of the money you could make from Pokémon

@DushyantPUNK This project is designed to show the flaws in 802.11b,g,n NOT To encourage people to disrupt other people's networks.

But I wouldn't be surprised to see an ESP8266 and this software being used in the next season of Mr Robot

spacehuhn, 2017-03-03:
MITM is a totally other thing. It is an **attack** to sniff traffic, you can do that with an raspberry but not with a $2 WiFi chip.

@poslogic ooooh it would be so awesome to see that on mr robot!! Someone should tell them this project exists!

poslogic, 2017-03-03:
I thought it was me!!!

Seems @DushyantPUNK Is editing things???

For context to anyone else reading this before the edit and user name change:

Add Password grabber for beacon span so as to get the password of the ap that we have deauthed. This will be helpful for hacking wifi without any attacks.

Correct me please if I am wrong.

poslogic, 2017-03-03:
I thought it was me!!!

Seems @DushyantPUNK Is editing things???

For context to anyone else reading this before the edit and user name change:

Add Password grabber for beacon span so as to get the password of the ap that we have deauthed. This will be helpful for hacking wifi without any attacks.

:Correct me please if I am wrong.

joneroy, 2017-03-03:
@spacehuhn MITM crazy idea really it's not possible with $2 WiFi chip with small memory.
@poslogic Mr.Robot idea is cool it's possible Elliot Alderson flashing esp8266-deauther into wemos mini pro useing small otg adopter and Darlene Alderson she wemos mini pro like a necklace. i think i contact with mr robot team an i'm saying them $2 WiFi and Stefan Kremser open source project esp8266_deauther hack any 2.4gh wifi.

closed 2017-03-03 by spacehuhn

anudeepND, 2017-03-04:
> Can we perform a mitm with this to sniff the traffic or to get the password of the ap?

Give this man a medal!

ITCMD, 2019-06-12:
I dont see how this would be too hard to implement? Just have an option to keep track of what passwords people attempt to enter to a beacon. Or is the problem that it doesnt actually recieved data, only send out empty packets?