SpacehuhnTech/esp8266_deauther #67: Scan for clients issue

keviiin38, 2017-03-03:
Hey guys, thanks for this great project first.

I'm using this [http://robotdyn.com/catalog/boards/wifi_nodem_esp8266_32m_flash/](url) card and followed all the instalation part, I'm able to scan for the WiFi networks (not 5Ghz) but when I select one and go to the "Clients" section, I'm not able to catch any client (I tried with different delay)

(Other question, i saw in some screenshots in other issues, a "settings" part, is it included some where or only someone's work ?)

Thanks in advance

spacehuhn, 2017-03-03:
Yes have a look at the testing branch: https://github.com/spacehuhn/esp8266_deauther/tree/testing
Try this version out there are a lot of improvments in it.  
Also be sure that you have a client device connected to the selected network wich produces traffic because otherwise the WiFi chip won't see packets and don't find your device even if it's connected.

joneroy, 2017-03-03:
@keviiin38  
# **Same testing branch prob #42**

keviiin38, 2017-03-04:
@spacehuhn 
Thanks very much I didn't see this branch and post !
It could be 5 devices but if they didn't send or receive any packages, I won't see them, exactly ?
I'll try again.

Thankss

closed 2017-03-04 by keviiin38