SpacehuhnTech/esp8266_deauther #68: Testing Branch AP not connecting

Robiii97, 2017-03-03:
When i change the SSID and Password in settings and reboot the ESP, my connection fails and im unable to connect to the ESP, but it does show the Acces point online with the SSID i changed.
I also tried changing the SSID and password in adruino before uploading but that gives me the same crashing result.

joneroy, 2017-03-03:
# **Same prob but solved** 
after change the SSID and Password in settings press **reset button on esp8266 module** and done.

[image]

Robiii97, 2017-03-03:
It still wont connect....

Robiii97, 2017-03-03:
Actually after some tries thad fixed it thanks!

poslogic, 2017-03-03:
There are extra steps in the testing branch!

https://github.com/spacehuhn/esp8266_deauther/blob/testing/README.md

13-18 are important.

Hope this helps.

joneroy, 2017-03-03:
@poslogic you found any prob

poslogic, 2017-03-03:
No only tested my NodeMCU board so far.
All good.

Will try testing on Wemos D1 mini & mini Pro over weekend.

All good so far and only getting better.

What about you?

poslogic, 2017-03-04:
@Robiii97 please close if your issue is solved.
Thank you :)

Please add any bugs or comments to this thread...https://github.com/spacehuhn/esp8266_deauther/issues/42

joneroy, 2017-03-04:
@poslogic  I found only two bugs. all working fine.

poslogic, 2017-03-04:
I wished that I was as clever as Stefan!

I would back him to be very successful in the future :100:

joneroy, 2017-03-04:
@poslogic i think teamwork and unity need for any work

poslogic, 2017-03-04:
Keep learning like I do.
But when you find (sorry Stefen) someone younger and more capable (intelligent) than yourself you have to be humble, help out and try to learn what you can from them.

Thanks for Teaching my (47 year old) new things.

joneroy, 2017-03-04:
@poslogic :100:

poslogic, 2017-03-04:
Happy times in the future for you as well, be good and good things happen!

anudeepND, 2017-03-04:
> Will try testing on Wemos D1 mini & mini Pro over weekend.
Hmmm, wemos D1 mini pro has an external antenna connector, range will be pretty good I think...

joneroy, 2017-03-04:
@poslogic thank you

joneroy, 2017-03-04:
@anudeepND Ya Wemos D1 mini pro good feature

Wemos D1 mini pro:

16M bytes(128M bit) Flash

External antenna connector

Built-in ceramic antenna

New CP2104 USB-TO-UART IC

Same size as D1 mini, but more light.

spacehuhn, 2017-03-04:
> I wished that I was as clever as Stefan!
👍 this issue got a bit offtopic I think ^^ 

> I would back him (invest money that he is) to be very successful in the future 💯
I'll come back to this later :D 

The Wemos d1 mini pro is an excellent board, but don't think the antenna connector will give you any noticeable advantage over the on board antenna. I've tested this already,

joneroy, 2017-03-04:
@spacehuhn Ya this issue got a bit offtopic :smile:

closed 2017-03-04 by Robiii97