SpacehuhnTech/esp8266_deauther #69: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'

tobozo, 2017-03-03:
Just cloned the testing version and got this error while trying to conpile : 

```
sketch/APScan.cpp: In member function 'bool APScan::start()':
APScan.cpp:14: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'
     results = WiFi.scanNetworks(false, true); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)
                                            ^
sketch/APScan.cpp:14:44: note: candidate is:
In file included from sketch/APScan.h:6:0,
                 from sketch/APScan.cpp:1:
/home/tobozo/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)
     int8_t scanNetworks(bool async = false);
            ^
/home/tobozo/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided
exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'

```

Changin `results = WiFi.scanNetworks(false, true);` by `results = WiFi.scanNetworks();` seems to fix the problem but I've got a feeling this is going backwards in terms of features.

Do I need to change something to my Wifi library in order to get this working ? I thought Async Wifi scan only happened in an earlier version of the SDK and was not available yet in the 2.0.0

tobozo, 2017-03-03:
ooops sorry closing this and re-posting in the correct thread

closed 2017-03-03 by tobozo

spacehuhn, 2017-03-03:
Yes you're right it's the Wi-Fi library. Follow the steps in the readme. It has changed in the testing branch because of the hidden AP scan.

dxsakthisundar1, 2019-10-15:
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)' 
how to solve this .. can anyone explain please