SpacehuhnTech/esp8266_deauther #72: Error: Setting box values blank

joneroy, 2017-03-04:
Setting box values blank it's big prob please solve this issue

[image]

spacehuhn, 2017-03-04:
like I said many times before: look at the logs and use the dev tools of your browser. Otherwise we can't help you.

It's like you ask why your car isn't starting and sending a picture of it. I have no idea...

joneroy, 2017-03-04:
@spacehuhn can i send you serial monitor results.

joneroy, 2017-03-04:
# **Bugs:**

SSID: pwned
SSID length: 5
password: deauther
password length: 8
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
attack SSID encrypted: 0

starting...
starting WiFi AP
SSID: pwned
Password: deauther
generating Macs...done
getting AP scan result JSON { "aps":[ ] }
done
getting settings json
{"ssid":"pwned","password":"deauther","deauthReason":1,"attackTimeout":300,"attackPacketRate":10,"clientScanTime":15,"attackEncrypted":0,"nameList":[{"n":"null              ","m":"88:70:8c:a3:cd:0f","v":"LenovoMo"},{"n":"
                                                                   wbChin","m":"20:20:20:20:20:20","v":""},{"n":"                  ","m":"61:20:20:20:20:20","v":""},{"n":"","m":"20:20:20:20:00:00","v":""}] }
done
settings:
SSID: jone doe
SSID length: 8
password: 123456789
password length: 9
deauth reson: 0
attack timeout: 0
attack packet rate: 0
client scan time: 0
attack SSID encrypted: 0
settings saved
getting settings json
{"ssid":"jone doe","password":"123456789","deauthReason":0,"attackTimeout":0,"attackPacketRate":0,"clientScanTime":0,"attackEncrypted":0,"nameList":[{"n":"null              ","m":"88:70:8c:a3:cd:0f","v":"LenovoMo"},{"n":"
                                                                   wbChin","m":"20:20:20:20:20:20","v":""},{"n":"                  ","m":"61:20:20:20:20:20","v":""},{"n":"","m":"20:20:20:20:00:00","v":""}] }
done
getting settings json
{"ssid":"jone doe","password":"123456789","deauthReason":0,"attackTimeout":0,"attackPacketRate":0,"clientScanTime":0,"attackEncrypted":0,"nameList":[{"n":"null              ","m":"88:70:8c:a3:cd:0f","v":"LenovoMo"},{"n":"
                                                                   wbChin","m":"20:20:20:20:20:20","v":""},{"n":"                  ","m":"61:20:20:20:20:20","v":""},{"n":"","m":"20:20:20:20:00:00","v":""}] }
done

[image]

spacehuhn, 2017-03-04:
thank you! Try to clear the namelist -> at settings the last button 'clear'.

joneroy, 2017-03-04:
@spacehuhn 
Nothing showing in console.

[image]

spacehuhn, 2017-03-04:
bugfix: 2e5656cfba0d595e4f83e188d5e385991ee0d32c

closed 2017-03-04 by spacehuhn

joneroy, 2017-03-04:
@spacehuhn 

thank you!