SpacehuhnTech/esp8266_deauther #73: Can't upload to ESP8266

SamuelKlit, 2017-03-04:
When i try to upload it just gives me this error: 

>\System folders\OverfÃ¸rsler\esp8266_deauther-master\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino: In function 'void setup()':
esp8266_deauther:70: error: 'load404' was not declared in this scope
esp8266_deauther:72: error: 'loadIndex' was not declared in this scope
esp8266_deauther:74: error: 'loadClients' was not declared in this scope
esp8266_deauther:75: error: 'loadAttack' was not declared in this scope
esp8266_deauther:76: error: 'loadFunctionsJS' was not declared in this scope
esp8266_deauther:79: error: 'loadStyle' was not declared in this scope
esp8266_deauther:82: error: 'sendAPResults' was not declared in this scope
esp8266_deauther:83: error: 'startAPScan' was not declared in this scope
esp8266_deauther:84: error: 'selectAP' was not declared in this scope
esp8266_deauther:85: error: 'startClientScan' was not declared in this scope
esp8266_deauther:86: error: 'sendClientResults' was not declared in this scope
esp8266_deauther:87: error: 'selectClient' was not declared in this scope
esp8266_deauther:88: error: 'setClientName' was not declared in this scope
esp8266_deauther:89: error: 'sendAttackInfo' was not declared in this scope
esp8266_deauther:90: error: 'startAttack' was not declared in this scope
exit status 1
'load404' was not declared in this scope

I just cloned the project. I use the 2.0.0 SDK with an ESP-12E. And i also remembered to add the lines to user_interface.

SamuelKlit, 2017-03-04:
I seem to have fixed it. I moved the project folder to my desktop. I suspect it didn't work because the folder where it was saved in before had Ø in it.

closed 2017-03-04 by SamuelKlit

joneroy, 2017-03-04:
@SamuelKlit It's working right now