SpacehuhnTech/esp8266_deauther #75: ESP8266 with hidden SSID

anudeepND, 2017-03-04:
Is it still possible to deauthenticate with hidden SSID. The NodeMCU doesn't broadcast it's SSID.
Here's the code:
`void softAP(const char* ssid, const char* passphrase, int channel = 1, int ssid_hidden = 0); `

anudeepND, 2017-03-04:
I don't have a esp8266 right now, can anyone please confirm that it works or not?

joneroy, 2017-03-04:
@anudeepND Ya it's possible this feature is recently implemented to testing branch to esp8266_deauther

anudeepND, 2017-03-04:
I haven't been able to try it out.
It is about hiding our SSID, not be confused with deauthenticating hidden SSID of victim...

joneroy, 2017-03-04:
@anudeepND what mean of

> It is about hiding our SSID

joneroy, 2017-03-04:
https://github.com/spacehuhn/esp8266_deauther/tree/testing

anudeepND, 2017-03-04:
Consider like this, I have a Nodemcu with WiFi name as 'deauther', so when I turn my phone's WiFi on, I can see 'deauther' or anyone with WiFi can see the name. 
It is like not broadcasting the name 'deauther'

joneroy, 2017-03-04:
This feature is recently implemented to testing branch to esp8266_deauther

N0vaPixel, 2017-03-04:
You doesn't need the ssid to communicate with AP..

joneroy, 2017-03-04:
[image]

joneroy, 2017-03-04:
@anudeepND 

## **I hope you find your ansr so close this issue.**

anudeepND, 2017-03-04:
Apologies. As joneroy said, it is already included in testing branch. Sorry

closed 2017-03-04 by anudeepND