SpacehuhnTech/esp8266_deauther #77: Different Boards?

TKems, 2017-03-05:
I purchased a esp8266 from Aliexpress (China) and it arrived today. I watched Seytonic's video about it, and was eager to start. I plugged in the board and got to work on setting it up. Once I had gotten everything plugged in a working, I decided to upload it to the board. Here is a picture of the board that I have: 
[image]

[image]

I tried uploading to the board with the Generic ESP8266, but it failed with an error. (Sorry, I didn't copy it and don't have it to share.) 

After about 2 hours of messing with settings, code, and files, I looked at the comment section in Seytonic's video. Some people were having the same issue as I was. 

To clarify, I tried different boards, different CPU frequencies, different baud rates, etc. Nothing was working. 

I found out that for this board, pictured above, it uses these settings, which other people may find useful: 

Board: NodeMCU 0.9 (ESP-12 Module)
CPU Freq.: 80 MHz 
Upload using: Serial
Flash size: 4M (1M SPIFFS)
Upload speed: 115200

Also, you MUST hold down the flash button, reset the device, using the reset button, then flash it (you do not need to hold down the flash button while flashing the device, only when resetting it) 

I think that the FAQ and Readme may be improved if this is added. 

Where I purchased the board: https://www.aliexpress.com/item/V3-4-M-bytes-32-Mbits-FLASH-Lua-NodeMcu-placa-de-desarrollo-de-Redes-WIFI-Basado/32711244582.html

Image of comment from YouTube: 
[image]

joneroy, 2017-03-05:
@TKems 

same board i'm using i think you don't config arduino.

[image]

[image]

[image]

download esp8266-deauther

https://github.com/spacehuhn/esp8266_deauther

# **README**

https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md

joneroy, 2017-03-05:
@TKems 

# **issue**

[image]

[image]

joneroy, 2017-03-05:
# **I hope you find your ansr so close this issue.**

Splodgie, 2017-03-05:
@joneroy  READ HIS POST AGAIN, He has it working, He has the setting, He's Flashed the ESP, He's just letting people know what worked for him and advice for others.
(I hope you find your ansr so close this issue,   Who died and made you GOD).

joneroy, 2017-03-05:
@Splodgie  apologize i think i skip some line reading time sorry it's my bad.

spacehuhn, 2017-03-05:
Thanks for sharing that information.

I also have one of these chunky big "v3" boards and mine works fine with "NoceMCU 1.0 (ESP-12E)" and 115200 baud rate without pressing a button.

Yours should also work with this settings, it's clearly not the 0.9 NodeMCU version.
Maybe you need to install CH340G drivers? I'm not sure... my board comes from an other manufacturer maybe that's why.

I recommend the smaller Amica NodemMCU boards (like the one seytonic used) they use an other USB serial chip with is more reliable, esier and faster.

> I think that the FAQ and Readme may be improved if this is added.

Well yes but that's a gerneral problem between the user and their board and has nothing to do with this project.

Splodgie, 2017-03-05:
General Info.
Also, there are FAKE's out there, I know I have 3 of them.  The main faking part for me is the RAM chip, (you cannot go by what's printed on the Chip).  The ESP-12 could have 1M, ESP-12E should have 4M, that's what the change is from 12to12E, Ram size,
Check you ESP so you know How much RAM it has by loading the Ram check file to it, IDE/File/Examples/ESP266/CheclFlashConfig.  
Set the Setting for the Board as you beleive you have , load the file, open the 
serial Monitor, press reset on your ESP, It will show you what it's found about your RAM and settings, It reports the RAM size found,  and if your ESP setting are correctly set. (this is not a 100% perfect test).

https://en.wikipedia.org/wiki/ESP8266          .for more ESP Board Info.
http://www.voti.nl/blog/?p=52

spacehuhn, 2017-03-05:
Just for the sake of correctness, what you mean @Splodgie is the (external) flash size. The RAM is limited to 80kb on all models ;)

Splodgie, 2017-03-05:
Hi, Yes.  I should have said Flash Memory Chip  and Not RAM,

closed 2017-04-23 by spacehuhn