SpacehuhnTech/esp8266_deauther #78: My TODO List

N0vaPixel, 2017-03-05:
Hi !
I've created this issue because i don't want to make a new one for each idea i have.
So feel free to leave your questions, ideas and feedbacks.

**Ideas:**
* <del>Add a restart button in the web interface (soft reset the SDK)</del> **(done)**
* <del>Use built-in LED to see if an attack is running</del> **(done)**
* For ESP AP, add the abillity to switch between random bssid and chosen one.
* Enhance the deauth attack with new methods. ☠️ 
* Reduce the power consumption with an automatic idle mode. 🔋

spacehuhn, 2017-03-05:
Herr I have a question! *raising hand*
I don't get what you mean by switch between random and chosen bssid - what will be the difference?

Also what do you mean by enhance deauth attack?

joneroy, 2017-03-05:
@spacehuhn 

> I don't get what you mean by switch between "**random**" and chosen bssid

It's mean esp8266 change his own mac randomly.

> I don't get what you mean by switch between random and "**chosen**" bssid

It's mean you chosen bssid for esp8266.

spacehuhn, 2017-03-05:
What do you mean by "enhance deauth attack"?
And can you set it to sleep when it's in AP mode? I thought about that too but never tested it.

N0vaPixel, 2017-03-05:
**1_** @spacehuhn  Currently the esp_8266_deauther deauth only in "one way". It spoof ap and send deauth & disassociation to client. To ensure his effeciency in all cases, it would be nice to add the others way.
Them there are 4 types/flavor of "deauth/disassociation " packet:

Type 1: Send deauth from AP to client
Type 2: Send deauth from client to AP
Type 3: Send disassociation from AP to client
Type 4: Send disassociation from client to AP

You will be able to choose whose ones you want to use.
This is how famous tools like "mdk3" works. [http://svn.mdk3.aircrack-ng.org/mdk3/docs/Documentation_incomplete.html](url)

Easy to implement i'll do it later. I just need to modify the buildDeauth method to receive more parameters.

**2_** For the sleep mode, i'll see if it actualy work in an experimental way.
Why not reduce the tick of the loop function when no attack is running and no action has been made by the user since few minutes

H1ghTech, 2017-03-05:
Yep it would be good to reduce the tickspeed cause the other way won't work: you'd have to set ch_pd to low but then the esp is shut down so you would have to manually set ch_pd high

N0vaPixel, 2017-03-05:
@H1ghTech  Great idea. But i'm wondering if i power off the esp8266 module how the user can power on it without a physical access to the board

spacehuhn, 2017-03-05:
ok so first:
at the moment Type 1 and Type 3 is already in use and I wouln't not implement the other 2 because then it would be super easy for router to track you and block the attack.

about the sleep mode... correct me if I'm wrong but what I tested so far was that it doesn't matter what you do it will draw around 80mA. You can only reduce that by setting it into one of the sleep mode, for that you need to be able to wake it up, which I think is not possible without user interaction and at this point it's easier to just unplug the device if you don't use it.

N0vaPixel, 2017-03-05:
I didn't heard about AP that actively detect a spoofed packet, since many of the FAI AP don't let her customers implement scripts.
A skilled network admin can detect and track you as a skilled guy victim of a deauth attack.
Even more, a network admin is able to detect any deauth packets "in the air", affecting her clients or his AP.
I don't planned to force the use of the other way, they will be disabled by default. But can be enabled by user depending on the situation and his needs. Providing are more customizable tools.

Yes, i'm talking about sleep mode. What about implemeting in the end of loop() a function that calculate the sleep time.
The sleep time will increase if no action has been made. It'll reach a maximum time arround 5-10sec. So the first reconnection will take more time.
The goal is to increase his lifetime when you use it with a external battery, and you put it in hidden location.

spacehuhn, 2017-03-05:
It's not about the spoofed packets, it's about the number of packets. When you send them to the router it have to react to them and if you spam these packets (what the ESP does) the router will slow down or, if implemented, detect it as dos attack.

But the other way around only the clients you want to disconnect are affected and so you minimize the chance of beeing detected. Of course there are routers which monitor all packets, but these are expensive high end systems which are only used for big buildings. 99% of the home routers don't have that.

N0vaPixel, 2017-03-05:
@spacehuhn  Yes you right, thanks for your feedback

spacehuhn, 2017-03-05:
no problem and thanks for your work! 👍  I probably won't have time to work on this for the next weeks but let me know if you need any help :)

N0vaPixel, 2017-03-05:
@spacehuhn  I'll keep you up-to-date if i encounter problems 👍 
Love this project ❤️

joneroy, 2017-03-05:
@N0vaPixel 

built-in LED not working when attack is running

H1ghTech, 2017-03-05:
@joneroy got the same problem

joneroy, 2017-03-05:
@H1ghTech #87

N0vaPixel, 2017-03-06:
What about an information web page, containing AP mac, clients connected to AP ect... ?

anudeepND, 2017-03-07:
You have all those info on web interface.

N0vaPixel, 2017-03-07:
Oh by saying "AP" i mean the esp AP

spacehuhn, 2017-03-07:
Hey @N0vaPixel I added you as collaborator, you're doing a great job and I'm very thankfull for your help, as attation on this project grows and grows! 👍  
You can push directly into the testing branch, just don't merge it into the master before I looked over it. You can also close some of the useless issues that are posted everyday.

N0vaPixel, 2017-03-07:
A pleasure to collaborate with you, thanks @spacehuhn 👍 
I'll contact you if i think that the testing branch is stable and can be merged

N0vaPixel, 2017-03-08:
**Update :** Now you can see which AP a client is associated with. Report if you encounter any bug. Thanks :)

spacehuhn, 2017-03-11:
Hey @N0vaPixel do you think we can make a new release? I couldn't find a new bug so far. 
I know there are not many changes, but the js compatibility is probably a big issue for many people and now that it's fixed (thx for that btw!) I think a version 1.1 would be appropriate.

spacehuhn, 2017-03-12:
I'm sorry if I am a bit to fast with the merging, I just wanna get this done: https://github.com/spacehuhn/esp8266_deauther/pull/113

Thank you again for your work 👍 and I hope we see less issues due to this release :)

N0vaPixel, 2017-03-12:
Hi @spacehuhn
Sorry i'm very bust this week-end. 
A new release is a good thing, i didn't noticed any kind of bug. 
I plan to commit the mac changer feature next week. But before i need to fix some bugs

tpmodding, 2017-03-12:
saw this yesterday, dont know if its usefull for you guys
http://hackaday.com/2017/03/10/save-esp8266-ram-with-progmem/

spacehuhn, 2017-03-12:
@tpmodding I used the progmem for all the html files and the mac vendor list. Just have a look at the data.h ;)

tpmodding, 2017-03-12:
i'm just learning c...i'm new in this coding thing... :D this is to high for me :P

N0vaPixel, 2017-03-12:
@tpmodding Thx for your help
But we already use PROGMEN to store all web pages

spacehuhn, 2017-03-12:
> EDIT: I think mac changer may be the last feature that we can add in the settings web page, the esp tends to run out of memory 😢

You mean because the html file gets to big? We could split the JS part into a seperate file.

N0vaPixel, 2017-03-12:
I have splitted the JS part, we will see if it helps. Also, url are now more sexy ❤️  (/attack.html is now /attack)
I would be nice to make a python script that auto minify and auto update data.h 👍

cezarlacatus, 2017-03-28:
Capturing handshake maybe??

cezarlacatus, 2017-03-28:
Usb rubber duckey??It would be a great ideea

DrHyperKALICH, 2017-03-28:
Hello! My idea:

Please add the

deathmode = true;

Then esp8266 on boot starts to scan wifi network with a period of 1 minute (for example) and deauth all ssid/client.

But add the ability to make a list with ssid exeptions for this action.

So once enabled, it must start work on boot, with no need to go to 192.168.4.1 to make actions.

cezarlacatus, 2017-03-28:
Wifi reapeater..

spacehuhn, 2017-03-28:
@cezarlacatus that's not what this project is for

hallard, 2017-04-09:
Guys,
This is a very nice project, I love it ;)
Anyway may be I can help with some ideas 
- first I can implement asyncwebserver from @me-no-dev which is a wonderful library that speed up web server and it's becoming the reference because it's async and no blocking
- use the spiffs file system to serve HTML and JS files 
- put the MAC address list on Spiffs to get back flash memory for code 
- add web file spiffs editor to be able to edit spiffs files from web interface (great for testing)

Let me know if you want me to help with pull request or as collaborator on a new branch, whatever is fine for me but first implementation of async need lot of changes ;)

WillGreen98, 2017-04-09:
I am using Atom with Platformio for any Arduino projects; I managed to upload the project to my NodeMCU, however, it has messed up, any ideas of how to wipe it with esptools or something?

Also, during compilation, the gitignore is not functioning as needed as .pioenvs are being compiled and are causing errors, what should be done with this? 

Thanks :)

Mcewen123, 2017-04-10:
> I am using Atom with Platformio for any Arduino projects; I managed to upload the project to my NodeMCU, however, it has messed up, any ideas of how to wipe it with esptools or something?

> 
[image]

BananaMolester, 2017-04-11:
Would adding an option to enable not password protected beacon(list so people would get a notification on their phone when we broadcast it?

BingeOn, 2017-04-18:
Is there any way it could work on 5ghz or maybe there's another board that could do this at 5ghz?

FrancTheCat, 2017-04-23:
I would like to equip my jammer with a list of SSIDs from a file just like the Beakon spamer from Spacehuhn!

jLynx, 2017-04-23:
@FrancTheCat its not a jammer

FrancTheCat, 2017-04-23:
Better? 😉

amahlaka, 2017-04-28:
Suggestion: support for WebApp / Add to homescreen (https://developer.chrome.com/multidevice/android/installtohomescreen)
I have already started to implement this on my fork

jLynx, 2017-08-09:
@spacehuhn can this be closed now?

spacehuhn, 2017-08-09:
...  that is @N0vaPixel decision.

Behooz-Sirouszad, 2017-09-24:
connection and disconnection problem for user to work with html pages ...iv found the problem and edited yout data.h code  buffer size from 6000 must be change to 1000 and all works fine und wundabar.
vielen dank Fur ihre aufmerksamkeit..

spacehuhn, 2018-03-24:
I think it's fair to say that we can close this.

closed 2018-03-24 by spacehuhn