SpacehuhnTech/esp8266_deauther #81: Deauthentication Attack Using lua Script

joneroy, 2017-03-05:
it's wifi deauthentication attack esp8266 using lua script.

H1ghTech, 2017-03-05:
Yup it's possible but very inefficient

joneroy, 2017-03-05:
Why inefficient ?

H1ghTech, 2017-03-05:
Cause it doesn't attack very fast.
,You can't kick anyone out because it's to slow

joneroy, 2017-03-05:
@H1ghTech and NodeMcu frimware it's a elua based frimware, 

> NodeMcu

Advanced API for hardware IO, which can dramatically reduce the redundant work for configuring and manipulating hardware. Code like arduino, but interactively in Lua script.

joneroy, 2017-03-05:
@H1ghTech and What about Micropython now it's available for esp8266

joneroy, 2017-03-05:
@H1ghTech 

> Cause it doesn't attack very fast.
You can't kick anyone out because it's to slow.

I think it's impossible with lua but now no esp8266_deauther source code available.

H1ghTech, 2017-03-05:
I think micropython would work great

joneroy, 2017-03-05:
@H1ghTech i think micropython better then lua

H1ghTech, 2017-03-05:
That's what I just said lol

joneroy, 2017-03-05:
@H1ghTech You right :100:

spacehuhn, 2017-03-05:
@joneroy stop posting useless messages like this one in every damn issue! it's really annoying and I'll delete them.

spacehuhn, 2017-03-05:
I don't meant the issue, I meant your messages like this one:

> @H1ghTech You right 💯

closed 2017-03-11 by spacehuhn

ebad84, 2023-09-24:
is there any way to do it on micropython?