SpacehuhnTech/esp8266_deauther #82: Broken ESP boards

spacehuhn, 2017-03-05:
Hey so I think my NodeMCU just broke.
I'm not sure though... 

It still open ups the access point and if I get connection (50/50 chance) I can't open up the webinterface. The same result with the examples and other versions of the deauther code.
I think it's a hardware problem because my other boards are working fine.

But my wemos board had an other issue a few days ago where I have to desolder the flash chip and erase it manually. After I soldered it back on it worked fine and it still does. Maybe I have to do the same with my NodeMCU?

Splodgie, 2017-03-05:
Hi, Sorry me again, I had a similar problem. The way I got round my problem was to use the NodeMCU Firmware Programmer, and Blank.bin all aeras of the flash chip ( I'm going to remove this post if it's wrong )

spacehuhn, 2017-03-05:
yes I tried that too, but with no success :(
I think my board is just dead. 😢

joneroy, 2017-03-05:
@spacehuhn I'm using NodeMCU frimware but my board not dead.

N0vaPixel, 2017-03-05:
@spacehuhn  Did your nodeMCU return blank web pages ? This is happening to me

spacehuhn, 2017-03-05:
nope I can't even connect and if I only get a 404 or timeout error and this happens not only with this project.

Splodgie, 2017-03-05:
I had Several Fake ESP(512k) and 6x 1MB -01 versions, so I changed all my ESP's to 4MB. (kept one -01 at 1M for testing)
10 Chips http://www.ebay.com/itm/10pcs-W25Q32FVSSIG-W25Q32FVSIG-25Q32FVSIG-4M-Memory-Flash-SOP-8-SMD-/201098254024?hash=item2ed263d6c8:g:TxkAAOSw9N1VrfC9.

(Power Problem)

spacehuhn, 2017-03-05:
Keep in mind that only the ESP-12E,F,S comes with 4mb. All other versions come with 1mb and the old versions like the original ESP-01 and ESP-12 only come with 512kb. So these are not fake. 

I don't think there are any fake board out there, they all use the ESP8266 chip and which other components the manufacturer use is up to them.

joneroy, 2017-03-05:
chip changing is possible "ESP-12" it's covered with metal sheet

[image]

Splodgie, 2017-03-05:
When a ESP-12E is sold as 4M, And when you unsolder the shield and find the chip has the marking scraped of, and it only responds using 512K settings, I call it Fake. 
The first 4MB chip I got of eBay Marked as 4MB, when fitted is 1MB I call that Fake.
My ESP-12 had 1MB as the eBay listing stated. some claim to have 4MB. and are on the 12 PCB (Only 2 sets of pinouts, not the 3 sets of the 12E)
( I agree External components are at the manufacturers digression, But when it's under a Shield and part of a sealed module on a PCB, add dose not meet the rating it should be, it's Fake).

spacehuhn, 2017-03-05:
hm I guess you're right then.

All boards and single ESP8266's I bought so far (all directly from china) where fine 😕 
But I also bought 10 of the 4mb EEPROM chips just to be safe.

Splodgie, 2017-03-05:
Where did you get the chips from ? (Some of the winbond chips are in a 1.2mm larger package and you have to bend the legs to make them fit,)

joneroy, 2017-03-05:
Anyone use Cloud-Linked ESP-100

spacehuhn, 2017-03-06:
Hey so I just got a new NodeMCU but the v0.9 devkit version. It's beautiful ❤️ 
It looked pretty big on the picture but it's actually smaller then the "v3" NodeMCU and it has a suprisingly good quality! Here is it with my lolin board:
[image]

I think I might buy another one of these. The other NodeMCUs that are out there come from very different vendors. The chance is pretty good that the one I had before (which broke) were just a cheap fake but this old version seem very good - and they maybe fake safe because nobody would sell there boards as v0.9 right?

Splodgie, 2017-03-07:
Hi, The bad point about these boards is there not breadboard friendly, I end up using 2 small BB's  to use them. 
I've just sent for 2 ESP-03 for the DeAuther project, to make it as small as possible. I know the 03 is not the smallest ESP,but it's 2.4GHz antenna needing to be 30mm Long and hiding it. 
I'll see how it goes when they turn up (and what turns up) If it looks any good I'll post some pics .

spacehuhn, 2017-03-07:
yea that's true they are really not breadboard friendly!
and I also wanted to build it as tiny as possible but I can't get my esp-03 to work.. but I bought a few of the new ESP8285s and I can't wait to try them out!

Splodgie, 2017-03-07:
Hi,You probable are, but I'll ask anyway,You are putting GPI0 to CD_PD, (I forgot to do this once after making 3 CaptiveIntraweb throwies, incasing them in epoxy resin,).

spacehuhn, 2017-03-08:
So because @Splodgie spontaneously decided to delete all his posts, this conversation doesn't make any sense anymore. Thanks for that!

closed 2017-03-08 by spacehuhn