SpacehuhnTech/esp8266_deauther #83: Bug with web server.

N0vaPixel, 2017-03-05:
There is a memory probelm. I noticed if a web page size around 7000 bytes, when you start an attack and try to load the page, the web server will return a blank page.

joneroy, 2017-03-05:
@N0vaPixel 

> the web server will return a blank page

This is error 404 blank page

N0vaPixel, 2017-03-05:
@joneroy  No, nothing is returned, this mean no HTTP header and body. It's not an error 404

joneroy, 2017-03-05:
@N0vaPixel 

I think it's a bug

N0vaPixel, 2017-03-05:
I minfied all the web pages and the "bug" disapeared. This prove that the memory usage is at stake.

spacehuhn, 2017-03-05:
Yeah that's not really a bug, at some point you can't implement new things because the RAM is full. Do you use my minifier.html in the htmlfiles folder? I minify them first and then turn them into bytes.

N0vaPixel, 2017-03-05:
@spacehuhn Yes it's really useful, i decided to use it.

joneroy, 2017-03-05:
@N0vaPixel 

minifier.html is very useful for update the html data.h

N0vaPixel, 2017-03-05:
Yes i know, i wouldn't be able to update data without byte-ify 😸

closed 2017-03-05 by N0vaPixel